Xilinx/xfopencv #2: Is there problem with using xf::Threshold()?

kimfeng, 2017-11-16:
I use xf::Threshold() to Instead opencv cvThreshold and cv::threshold. when in class SimpleBlobDetector:
_
void SimpleBlobDetector::detectImpl(const cv::Mat& image, std::vector<cv::KeyPoint>& keypoints, const cv::Mat&) const
{
...
//params.minThreshold = 200;
for (double thresh = params.minThreshold; thresh < params.maxThreshold; thresh += params.thresholdStep)
{
Mat binarizedImage;
//cv::threshold(grayscaleImage, binarizedImage, thresh, 255, THRESH_BINARY);
xf::Mat<XF_8UC1, HEIGHT, WIDTH, NPIX> imgInput(grayscaleImage.rows, grayscaleImage.cols);
xf::Mat<XF_8UC1, HEIGHT, WIDTH, NPIX> imgOutput(grayscaleImage.rows, grayscaleImage.cols);
Mat accelMat(grayscaleImage.rows, grayscaleImage.cols, grayscaleImage.depth());
imgInput.copyTo(grayscaleImage.data);
threshold_accel(imgInput, imgOutput, (int)thresh, 255, THRESH_BINARY);
accelMat.data = imgOutput.copyFrom();
binarizedImage=accelMat.clone();
_
I can get the same result as I got from opencv 2.4.5. But when I use it at the end of function cvStartFindContours() in contours.cpp
_
CV_IMPL CvContourScanner cvStartFindContours( void* _img, CvMemStorage* storage, int  header_size, int mode, int  method, CvPoint offset )
{
...
/* converts all pixels to 0 or 1 */
if( CV_MAT_TYPE(mat->type) != CV_32S )
{
//cvThreshold( mat, mat, 0, 1, CV_THRESH_BINARY );
xf::Mat<XF_8UC1, HEIGHT, WIDTH, NPIX> imgInput (mat->rows, mat->cols);
xf::Mat<XF_8UC1, HEIGHT, WIDTH, NPIX> imgOutput(mat->rows, mat->cols);
Mat accelMat((const CvMat *)mat), dst0=accelMat;
imgInput.copyTo(accelMat.data);
threshold_accel(imgInput, imgOutput, 0, 1, THRESH_BINARY);
accelMat.data = imgOutput.copyFrom();
accelMat.convertTo(dst0, dst0.depth());
}
return scanner;
}
_
It can not get the same result as i got result when I using opencv
Did I do something wrong? And how can I fix it?

bgouthamb, 2018-02-06:
In _xf::Threshold_, the _max_value_  is not taken as an argument, instead, it's value is fixed to 255. 

In your _SimpleBlobDetector_ function, the intended _max_value_ is 255 and the fixed value is also 255. Hence, it matches with OpenCV result. In _cvStartFindContours_ function, the intended value is 1 which differs from the fixed value 255. Hence, the difference in result.

The xf::Threshold function will be updated soon, to match OpenCV's cv::threshold.

bgouthamb, 2018-12-21:
xf::Threshold now matches OpenCV Threshold  function.

closed 2018-12-21 by bgouthamb