Xilinx/xfopencv #6: reVISION stack

Dominatrix007, 2018-05-24:
hello,
I was trying to execute an example from xfopencv design examples using reVISION platform zcu104. I am doing it through the terminal using the makefile. It works fine till the command comes "creating Vivado project and starting FPGA synthesis" after which it says synthesis is running and the PC gets hanged. After which I have to reboot my PC.
Alternatively I also tried doing it with sdx GUI and there also I am encountering the same problems. I tried executed some of the examples and in all of them I am facing this issue.
Please, help me resolve this issue.

 

****** vpl v2017.4 (64-bit)
  **** SW Build 2086221 on Fri Dec 15 20:54:30 MST 2017
    ** Copyright 1986-2017 Xilinx, Inc. All Rights Reserved.

INFO: [VPL 60-895]    Target platform: /home/ignko-train/zcu104_rv_ss/zcu104_rv_ss.xpfm
INFO: [VPL 60-423]   Target device: zcu104_rv_ss
INFO: [VPL 60-251]   Hardware accelerator integration...
Creating Vivado project and starting FPGA synthesis.
[14:08:05] Synthesis is running.
[14:09:05] Synthesis is running.
[14:10:07] Synthesis is running.
[14:11:13] Synthesis is running.
[14:12:14] Synthesis is running.
[14:13:20] Synthesis is running.
.....

Thank you!
Anjali

bgouthamb, 2018-05-25:
@Dominatrix007  Could you let us know which function you tried to build, the SDx version and what OS are you using?

Dominatrix007, 2018-05-25:
I tried to build the accumulate example using the makefile. 
I am using the SDx version 2017.4 and my OS is Ubuntu 16.04

Thanks!

bgouthamb, 2018-05-25:
@Dominatrix007  It might be an 'Out Of Memory' issue. What is the RAM size of your system? Are you facing this issue when building for other platforms also? Can you try in a system that has a larger RAM?

Dominatrix007, 2018-05-28:
@bgouthamb  hello,
The memory size is 88 GB. I don't think its a memory issue because it has not given any error related to nonavailability of memory. 
Yes, I have tried it with other platforms also but there also I am getting the same problem. Currently, I don't have any system with a larger RAM. Can you tell me any alternative?

Thanks!

Anjali-iit8, 2018-05-30:
@bgouthamb 
Hi,
I tried running on other system with more larger RAM. I was able to execute it. Looks like it was the memory issue.
Thanks for the help!!

bgouthamb, 2018-05-30:
Good to know that it works. Closing the issue.

closed 2018-05-30 by bgouthamb