Xilinx/xfopencv #8: Typo in header file for canny detector?

wang120332076, 2018-06-01:
I am trying to compile Canny edge detector for my ZedBoard, and during compilation my SDx 2017.2 always reports this warning
```
..../inc/xfopencv/imgproc/xf_canny_utils.hpp:490:9: warning: unknown pragma ignored [-Wunknown-pragmas]
#pragma hls inline
        ^
```
I checked both master branch and tag 2017.2, finding that the "hls inline" are in lowercase in both branches/tags. Is that a typo or this is the right way to do it?

bgouthamb, 2018-06-06:
The lowercase or uppercase does not matter when synthesizing in Vivado HLS. 
Only in SDx, during the initial stage of compilation (of ARM), this warning is being issued, which anyway has no effect on the generated binary.

However, the pragmas in the code will be updated to use uppercase wherever required, to ensure no such warnings are thrown. Thanks for bringing this up!

bgouthamb, 2018-06-28:
Fixed

closed 2018-06-28 by bgouthamb