Xilinx/xfopencv #9: new

Anjali-iit8, 2018-06-04:
@bgouthamb 
Hi,
I want to implement a simple OpenCV code using the reVISION platform. Can you tell me how I can edit the Makefile so that I can implement my example also.

Thankyou

bgouthamb, 2018-06-06:
In the Makefile, for the _APPSOURCES_ variable, put all the files that are being used, as you do with any general compilation. Next,  you need to specify the function name, which you want to be implemented on FPGA, after the _-sds-hw_ command followed by the file name in which the function is being called. The _-files_ is for specifying the file in which definition of the function resides.

This is the syntax : 
-sds-pf <platform-name>  -sds-hw <function-name> <file-from-which-the-function-is-being-called> -files <file-in-which-the-function-definition-resides> -clkid <value> -sds-end

bgouthamb, 2018-06-06:
This is the syntax : 
-sds-pf <platform-name>  -sds-hw <function-name> <file-from-which-the-function-is-being-called> -files <file-in-which-the-function-definition-resides> -clkid <value> -sds-end

Anjali-iit8, 2018-06-06:
hi,
I did that but then also it was giving various errors. Also I didn't use any of the xfopencv functions.
I just used the standard opencv libraries.

Anjali-iit8, 2018-06-06:
hi,
these are the errors that I am getting.

Moving function main to Programmable Logic
ERROR: [SYNCHK 200-11] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/cvstd.hpp:622: Variable 'this.assign.0' has an unsynthesizable type 'i8P*' (possible cause(s): pointer to pointer or global pointer).
ERROR: [SYNCHK 200-72] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/cvstd.hpp:624: unsupported c/c++ library function 'strlen'.
ERROR: [SYNCHK 200-71] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/cvstd.hpp:625: function 'cv::String::allocate(unsigned long)' has no function body.
ERROR: [SYNCHK 200-22] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/cvstd.hpp:625: memory copy is not supported unless used on bus interface possible cause(s): non-static/non-constant local array with initialization).
ERROR: [SYNCHK 200-71] /home/ignko-train/xfopencv/examples/TYQi/image_registration.cpp:51: function 'cv::imread(cv::String const&, int)' has no function body.
ERROR: [SYNCHK 200-71] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/cvstd.hpp:667: function 'cv::String::deallocate()' has no function body.
ERROR: [SYNCHK 200-41] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/mat.inl.hpp:71: unsupported pointer reinterpretation from type 'Mat' to type 'i8*' on variable 'im'.
ERROR: [SYNCHK 200-71] /home/ignko-train/xfopencv/examples/TYQi/image_registration.cpp:133: function 'cv::imshow(cv::String const&, cv::_InputArray const&)' has no function body.
ERROR: [SYNCHK 200-71] /home/ignko-train/xfopencv/examples/TYQi/image_registration.cpp:135: function 'cv::waitKey(int)' has no function body.
ERROR: [SYNCHK 200-11] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/mat.inl.hpp:681: Variable 'im' has an unsynthesizable type 'Mat' (possible cause(s): structure variable cannot be decomposed due to (1) unsupported type conversion; (2) memory copy operation; (3) function pointer used in struct; (4) unsupported pointer comparison).
ERROR: [SYNCHK 200-42] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/mat.inl.hpp:681: pointer comparison is not supported.
ERROR: [SYNCHK 200-71] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/mat.inl.hpp:682: function 'cv::Mat::deallocate()' has no function body.
ERROR: [SYNCHK 200-61] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/mat.inl.hpp:686: unsupported memory access on variable 'im' which is (or contains) an array with unknown size at compile time.
ERROR: [SYNCHK 200-71] /home/ignko-train/zcu102_rv_ss/sw/sysroot/usr/include/opencv2/core/mat.inl.hpp:571: function 'cv::fastFree(void*)' has no function body.
INFO: [SYNCHK 200-10] 14 error(s), 0 warning(s).
ERROR: [HLS 200-70] Synthesizability check failed.
command 'ap_source' returned error code
    while executing
"source /home/ignko-train/xfopencv/examples/TYQi/_sds/vhls/main_run.tcl"
    invoked from within
"hls::main /home/ignko-train/xfopencv/examples/TYQi/_sds/vhls/main_run.tcl -l main_vivado_hls.log"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
INFO: [Common 17-206] Exiting vivado_hls at Wed Jun  6 10:22:38 2018...
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/home/ignko-train/Xilinx/Vivado/2017.4/bin/vivado_hls /home/ignko-train/xfopencv/examples/TYQi/_sds/vhls/main_run.tcl -l main_vivado_hls.log'
sds++ log file saved as /home/ignko-train/xfopencv/examples/TYQi/_sds/reports/sds_image_registration.log
ERROR: [SdsCompiler 83-5004] Build failed

Makefile:28: recipe for target 'image_registration.o' failed
make: *** [image_registration.o] Error 1

bgouthamb, 2018-06-06:
Okay. I think you got the concept wrong here.  You cannot implement any kind of code on the SDx. The code has to be written in accordance with HLS & SDx(see UG902, UG1027 docs).  Go through any of HLS and SDx examples to get the idea.

Please post in the Xilinx forums for any further guidance.

closed 2018-06-28 by bgouthamb