Xilinx/xfopencv #10: Problem with xf_sw_utils.h file

Anjali-iit8, 2018-06-25:
Hi I am trying to port Canny edge detection function in my code for hardware acceleration. I am getting errors related to xf_sw_utils.h header file.

In file included from /home/ml1/workspace/image_registration/src/image_registration.cpp:12:
/home/ml1/zcu104_rv_ss/sw/a53_linux/inc/xfopencv/common/xf_sw_utils.h:120:98: error: expected '>'
     int cv_pix = cv_img.at<ap_uint<PixelType<DataType<_PTYPE,_NPC>::pixeldepth>::bitdepth>>(i,j);

I also checked this line in the header file many times, but the line there is different from what is specified in the error. The line is something like this:
int cv_pix = cv_img.at<ap_uint<XF_PIXELDEPTH ( XF_DEPTH (_PTYPE, _NPC))>>(i,j);

bgouthamb, 2018-06-25:
@Anjali-iit8 Unable to replicate this error. Can you please let me know these details: OS, SDx version, Platform, along with the flags and paths used in Makefile? 
Are you facing this issue when building the function _Canny_ standalone also (individual build, not as a part of your project)?

bgouthamb, 2018-06-25:
@Anjali-iit8 
XF_PIXELDEPTH(...), XF_DEPTH(...) are macros which resolve to the corresponding template specialization calls (see include/common/xf_types.h). The line in your error log file:

`int cv_pix = cv_img.at<ap_uint<PixelType<DataType<_PTYPE,_NPC>::pixeldepth>::bitdepth>>(i,j);`

 is the expanded form of the macros in the original code:

`int cv_pix = cv_img.at<ap_uint<XF_PIXELDEPTH ( XF_DEPTH (_PTYPE, _NPC))>>(i,j);`

Anjali-iit8, 2018-06-26:
@bgouthamb  OS: Ubuntu 16.04
SDx version: 2017.4
Platform: zcu104
I am building it using the GUI. I have build the Canny standalone and it is working but when I am building it as a part of my project then it is giving errors. I have attached the header file where I have defined the function.
[canny_edge.txt](https://github.com/Xilinx/xfopencv/files/2135509/canny_edge.txt)

and here in my code I have called the function:
 cv::Mat imgInput1;
	     imgInput1 = cv::imread(argv[1],0);
	     xf::Mat<XF_8UC1, 640, 480, XF_NPPC1> imgInput(imgInput1.rows, imgInput1.cols);
	     imgInput = xf::imread<XF_8UC1, 640, 480, XF_NPPC1>(argv[1],0);

	     cann(imgInput);

Also, I am able to port it to the PS side. This error is encountered when porting to PL side.

Anjali-iit8, 2018-06-26:
@bgouthamb 
I just checked, when I am running the Canny standalone using Makefile provided then it is working but when I am building it using SDx GUI then it is giving errors. So, I think problem is with working on GUI. Is there any extra steps required for building it using GUI?
I have included the necessary libraries and header files which were specified in the Makefile.

Anjali-iit8, 2018-06-26:
@bgouthamb 
Yes I have followed that process only, but then also I am getting errors while building. This is the error I am getting:
/home/ml1/workspace/canny/src/xf_canny_accel.cpp:43:13: error: array type 'typename DataType<8, 32>::name [64800]' is not assignable
 _dst1.data = (ap_uint<64>*)_dst.data;

bgouthamb, 2018-06-26:
@Anjali-iit8 
This is a different error you have posted now. Are you seeing this in the standalone build?  If yes, please check whether you have marked "xf::Canny<...>" and "xf::EdgeTracing<...>" to move to hardware, not "canny_accel".

Anjali-iit8, 2018-06-26:
@bgouthamb 
When I marked the above functions for acceleration then I am not getting this error but still I am not able to build:
make: *** No rule to make target 'xf_canny_accel.o', needed by 'canny.elf'.  Stop.

Also when I gave xf::Canny<...> function for the hardware acceleration in my code I am getting the same errors:
/home/ml1/zcu104_rv_ss/sw/a53_linux/inc/xfopencv/common/xf_sw_utils.h:120:98: error: expected '>'
     int cv_pix = cv_img.at<ap_uint<PixelType<DataType<_PTYPE,_NPC>::pixeldepth>::bitdepth>>(i,j);

bgouthamb, 2018-06-26:
@Anjali-iit8 
If it is building with Makefile and not building with GUI, might be that you are missing setting up something(?). I just had it built in my system in GUI with no issues.  Recheck your settings and try doing a clean build or new SDx project.

Anjali-iit8, 2018-06-26:
@bgouthamb 
Thank you. The Canny standalone is building now with the GUI. 
I also made a new SDx project of my own project also, and I gave the function xf::Canny<..> for hardware acceleration but when I am building it, it is not accelerating the function. It is not giving any errors but it is not accelerating the above function also. It seems like it is porting the whole code to the PS side. Also when I am trying to open the hardware function with Vivado standalone, it is giving error:  No Vivado HLS project exist for the given hardware function.

Anjali-iit8, 2018-06-26:
@bgouthamb 
If I understand right, we cannot give our own function for hardware acceleration, even if it comprises of only xfopencv functions?

bgouthamb, 2018-06-27:
@Anjali-iit8 
Of course, you can! See the color detection example in UG1233. 
Not accelerating a function, even after marking for hardware, seems to be a problem with the SDx GUI. Please post in SDSoC forum about this. 
Now that, your original issue about _xf_sw_utils.h_ is resolved, can you please close this?

Anjali-iit8, 2018-06-27:
Yes. Thanks

closed 2018-06-27 by Anjali-iit8