Xilinx/xfopencv #11: Warp transformation

Anjali-iit8, 2018-06-28:


bgouthamb, 2018-06-28:
@Anjali-iit8 
Warpaffine and Warpperspective are memory mapped implementations, whereas Warptransform is a streaming implementation. Functionally both are similar. For pipeline applications, Warptransform is more suitable.

Anjali-iit8, 2018-06-28:
@bgouthamb 
Can I also perform inverse warpAffine and inverse warpPerspective using warptransform because I cannot see any parameter for specifying it, like in OpenCV we give the flag WARP_INVERSE_MAP

bgouthamb, 2018-06-28:
@Anjali-iit8 
Yes, you can. There is no flag to set, you yourself need to generate and provide the inversed transformation matrix instead.

Anjali-iit8, 2018-06-28:
Okay. Thankyou

bgouthamb, 2018-06-28:
@Anjali-iit8 
Please close the issue if you think your question is answered.

closed 2018-06-29 by Anjali-iit8

LIFM6, 2024-11-12:
Hello, may I ask how you encapsulated the warp view function into an IP core? Can you teach me?