Xilinx/xfopencv #12: xf_resize_up_area.hpp:84:3: error: no matching function for call to 'xfExtractPixels'

piotr-aldec, 2018-07-05:
Hello,

I've tried to use resize function for more then 1 channel. Therefore, I've modified file xf_resize_config.h in example by changing TYPE to XF_8UC2
#define TYPE XF_8UC2

After calling make I got the following error:
../../include/imgproc/xf_resize_up_area.hpp:84:3: error: no matching function for call to 'xfExtractPixels'
                xfExtractPixels<NPC,WORDWIDTH,XF_DEPTH(DEPTH,NPC)>(line0,D0[i],i<<XF_BITSHIFT(NPC));
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../include/imgproc/xf_resize_up_area.hpp:411:34: note: in instantiation of function template specialization 'ProcessBlockAreaUp<6, 1, 1, 2>' requested here
                        XF_TNAME(DEPTH,NPC) out_pix = ProcessBlockAreaUp<DEPTH,NPC,WORDWIDTH,PLANES>(Hoffset,Hweight,Yweight,D0,D1,block_start,i);
                                                      ^
../../include/imgproc/xf_resize.hpp:72:4: note: in instantiation of function template specialization 'xFResizeAreaUpScale<2160, 3840, 2, 6, 1, 1, 1080, 1920, 3840, 1920>' requested here
                        xFResizeAreaUpScale<SRC_ROWS,SRC_COLS,XF_CHANNELS(TYPE,NPC),TYPE,NPC,XF_WORDWIDTH(TYPE,NPC),   \
                        ^
/space_1/storage/GIT/xfopencv/examples/resize/xf_resize_accel.cpp:34:6: note: in instantiation of function template specialization 'xf::resize<2, 6, 2160, 3840, 1080, 1920, 1, 2>' requested here
        xf::resize <INTERPOLATION, TYPE, HEIGHT, WIDTH, NEWHEIGHT, NEWWIDTH, NPC1, MAXDOWNSCALE> (_src, _dst);
            ^
../../include/common/xf_utility.h:64:6: note: candidate function [with NPC = 1, WORDWIDTH = 1, PIXELDEPTH = 2] not viable: no known conversion from 'typename PixelType<6>::uname [4]' to 'typename PixelType<2>::name *' (aka 'ap_uint<16> *') for 1st argument
void xfExtractPixels(XF_PTNAME(PIXELDEPTH)* tmp_buf, XF_SNAME(WORDWIDTH) &val1, int pos)
     ^
../../include/imgproc/xf_resize_up_area.hpp:85:3: error: no matching function for call to 'xfExtractPixels'
                xfExtractPixels<NPC,WORDWIDTH,XF_DEPTH(DEPTH,NPC)>(line1,D1[i],i<<XF_BITSHIFT(NPC));
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../include/common/xf_utility.h:64:6: note: candidate function [with NPC = 1, WORDWIDTH = 1, PIXELDEPTH = 2] not viable: no known conversion from 'typename PixelType<6>::uname [4]' to 'typename PixelType<2>::name *' (aka 'ap_uint<16> *') for 1st argument
void xfExtractPixels(XF_PTNAME(PIXELDEPTH)* tmp_buf, XF_SNAME(WORDWIDTH) &val1, int pos)

Does xf::resize function support more then one channel?

Thanks,
Best Regards,
Piotr

bgouthamb, 2018-07-05:
@piotr-aldec 
Resize function supports only single channel (XF_8UC1). See page 145 of [UG1233](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_2/ug1233-xilinx-opencv-user-guide.pdf).

bgouthamb, 2018-12-21:
Multichannel images will be supported in the upcoming release.

closed 2018-12-21 by bgouthamb