Xilinx/xfopencv #13: xfopencv error in makefile

dhruvchamania, 2018-07-11:
using 2017.4, getting this error while making the file(included the code.  in [makefile] as well as the what we input in terminal) for the sobel detector. 

ERROR: [SYNCHK 200-79] Cannot find the top function 'w0_xf_Sobel' in the design. Possible causes are: (1) the top function name is misspelled; (2) the top function is nonexistent or declared as static.
ERROR: [HLS 200-70] Synthesizability check failed.
command 'ap_source' returned error code
    while executing
"source /home/kushalchaudhari/Projects/_project/SDSoC_platform/xfopencv/examples/sobelfilter/_sds/vhls/w0_xf_Sobel_run.tcl"
    invoked from within
"hls::main /home/kushalchaudhari/Projects/_project/SDSoC_platform/xfopencv/examples/sobelfilter/_sds/vhls/w0_xf_Sobel_run.tcl -l w0_xf_Sobel_viva..."
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
INFO: [Common 17-206] Exiting vivado_hls at Wed Jul 11 12:18:03 2018...
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling '/home/kushalchaudhari/Xilinx/Vivado/2017.4/bin/vivado_hls /home/kushalchaudhari/Projects/_project/SDSoC_platform/xfopencv/examples/sobelfilter/_sds/vhls/w0_xf_Sobel_run.tcl -l w0_xf_Sobel_vivado_hls.log'
sds++ log file saved as /home/kushalchaudhari/Projects/_project/SDSoC_platform/xfopencv/examples/sobelfilter/_sds/reports/sds_xf_sobel_accel.log
ERROR: [SdsCompiler 83-5004] Build failed
sds++ completed at Wed Jul 11 12:18:03 IST 2018

MAKEFILE

APPSOURCES = xf_sobel_tb.cpp xf_sobel_accel.cpp
EXECUTABLE = sobel.elf
VERBOSE = #-verbose
PLATFORM = /home/kushalchaudhari/Projects/_project/example_desing/zip_file/zcu102_rv_ss
SDSCC_FLAGS = -sds-pf ${PLATFORM} -std=c++11 ${VERBOSE} \
-sds-hw "xf::Sobel<0,3,0,1,1080, 1920,1>" xf_sobel_accel.cpp -files ../../include/imgproc/xf_sobel.hpp -clkid 2 -sds-end -dmclkid 2
#sobel<XF_BORDER_CONSTANT,XF_8UC1,XF_16UC1,HEIGHT, WIDTH,XF_NPPC8>(imgInput, imgOutputx,imgOutputy);

CC = sds++ ${SDSCC_FLAGS}

IDIRS = -I. -I${SYSROOT}/usr/include -I ../../include

LDIRS = --sysroot=${SYSROOT} -L=/lib -L=/usr/lib -Wl,-rpath-link=${SYSROOT}/lib,-rpath-link=${SYSROOT}/usr/lib

LLIBS = -lopencv_imgcodecs -lopencv_core -llzma -ltiff -lpng16 -lz -ljpeg -lopencv_imgproc -ldl -lrt -lwebp -lopencv_features2d -lopencv_flann 
#LLIBS = -lopencv_core -lopencv_imgproc -lopencv_imgcodecs 
#OLEVEL = -ggdb
CFLAGS = -O3 -Wall ${OLEVEL} -c ${IDIRS} -DHLS_NO_XIL_FPO_LIB
LFLAGS = ${OLEVEL} ${LDIRS} ${LLIBS}

SOURCES := $(patsubst %,./%,$(APPSOURCES))
OBJECTS := $(APPSOURCES:.cpp=.o)
DEPS := $(OBJECTS:.o=.d)
.PHONY: all
all: ${EXECUTABLE}
${EXECUTABLE}: ${OBJECTS}
	${CC} ${OBJECTS} ${LFLAGS} -o $@
-include ${DEPS}
%.o: %.cpp
	${CC} ${CFLAGS} $< -o $@
clean:
	${RM} ${EXECUTABLE} ${OBJECTS}
ultraclean: clean
	${RM} -rf _sds sd_card ${EXECUTABLE}.bit ${EXECUTABLE} ${OBJECTS}
copy:
	/bin/cp -prf ${XLNX_SRC_PATH}/sobelfilter/example .
	/bin/cp -prf ${XLNX_SRC_PATH}/sobelfilter/src .

TERMINAL
source /home/kushalchaudhari/Xilinx/Vivado/2017.4/settings64.sh

export SYSROOT=/home/kushalchaudhari/Projects/Airbus_project/example_desing/zip_file/zcu102_rv_ss/sw/sysroot/

Hope this helps for you to resolve the issue.

bgouthamb, 2018-07-11:
@dhruvchamania 
Please check whether you have changed the configuration through the macros in _xf_sobel_config.h_ and _xf_config_params.h_. If yes, you need to update the parameters in the hardware function in the Makefile accordingly.

dhruvchamania, 2018-07-11:
Thanks for the reply, had not updated parameters. Now synthesis has begin, will update and close issue as soon as we test on ZC102. 
Another doubt till then, like we use caffe with python in opencv, can the same implementation be done with xfopencv?. If yes any detailed guides avaliable to implement that.

varunsanthaseelan, 2018-07-13:
@dhruvchamania , in order for you to run Python on zcu102, you will need a platform with a OS that supports python on it. The platforms for zcu102 that Xilinx ships today do not have that capability. Having said that, I would strongly encourage you to check out the PYNQ board and the computer vision overlays that can  be used from within Python (https://github.com/Xilinx/PYNQ-ComputerVision).

dhruvchamania, 2018-07-13:
@varunsanthaseelan i mean does zcu102 has the capabilites to use C++ with caffe like we use caffe with python. They have provided us with some examples, https://github.com/Xilinx/CHaiDNN but no particular guide to make such neural network implementation on own.

closed 2018-07-13 by dhruvchamania