Xilinx/xfopencv #14: xfopencv in Vivado HLS 2017.4

Yidagithub1103, 2018-07-18:
I want to rebuild your colordetect_accel part in Vivado HLS 2017.4.
I have done the C-Simunation,it works successful,but when i want to do the C Synthesis,I meet some error like this:
D:/Xilinx/Vivado/2017.4/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:62:5: error: the parameter for this explicitly-defaulted copy constructor is const, but a member or base requires it to be non-const
    nested_exception(const nested_exception&) = default;
    ^
D:/Xilinx/Vivado/2017.4/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:64:23: error: the parameter for this explicitly-defaulted copy assignment operator is const, but a member or base requires it to be non-const
    nested_exception& operator=(const nested_exception&) = default;
                      ^
D:/Xilinx/Vivado/2017.4/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:77:28: error: exception specification in declaration does not match previous declaration
  inline nested_exception::~nested_exception() = default;
                           ^
D:/Xilinx/Vivado/2017.4/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:66:20: note: previous declaration is here
    inline virtual ~nested_exception();
                   ^
D:/Xilinx/Vivado/2017.4/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:122:61: error: redefinition of default argument
    __throw_with_nested(_Ex&& __ex, const nested_exception* = 0)
                                                            ^ ~
D:/Xilinx/Vivado/2017.4/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:110:56: note: previous definition is here
    __throw_with_nested(_Ex&&, const nested_exception* = 0)
                                                       ^ ~
5 errors generated.

bgouthamb, 2018-07-26:
@Yidagithub1103 
2017.4 xfOpenCV code version doesn't have support for standalone Vivado HLS. Please download the 2018.2 version and follow the steps mentioned in HLS use model [document](https://github.com/Xilinx/xfopencv/blob/master/HLS_Use_Model/HLS_UseModel_Usage_Doc.pdf) to perform Synthesis.

Yidagithub1103, 2018-07-27:
@bgouthamb 
Thanks!
I've already downloaded the latest edition Vivado HLS.
I follow my previous step to do the simulation,but i meet error like this:

`D:/Xilinx/Vivado/2018.2/bin/vivado_hls.bat F:/Vivadoproject/HLS/colordecete/solution1/csim.tcl
INFO: [HLS 200-10] Running 'D:/Xilinx/Vivado/2018.2/bin/unwrapped/win64.o/vivado_hls.exe'
INFO: [HLS 200-10] For user 'Yida' on host 'desktop-bs6h1af' (Windows NT_amd64 version 6.2) on Fri Jul 27 10:03:34 +0800 2018
INFO: [HLS 200-10] In directory 'F:/Vivadoproject/HLS'
INFO: [HLS 200-10] Opening project 'F:/Vivadoproject/HLS/colordecete'.
INFO: [HLS 200-10] Opening solution 'F:/Vivadoproject/HLS/colordecete/solution1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 10ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_tb.cpp in debug mode
   Compiling ../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_accel.cpp in debug mode
   Generating csim.exe
Makefile.rules:399: recipe for target 'csim.exe' failed
In file included from D:/Xilinx/Vivado/2018.2/include/floating_point_v7_0_bitacc_cmodel.h:143:0,
                 from D:/Xilinx/Vivado/2018.2/include/hls_fpo.h:166,
                 from D:/Xilinx/Vivado/2018.2/include/hls_half.h:58,
                 from D:/Xilinx/Vivado/2018.2/include/ap_int_sim.h:73,
                 from D:/Xilinx/Vivado/2018.2/include/ap_int.h:65,
                 from F:/OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_config.h:36,
                 from ../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_tb.cpp:32:
D:/Xilinx/Vivado/2018.2/include/gmp.h:62:0: warning: "__GMP_LIBGMP_DLL" redefined
 #define __GMP_LIBGMP_DLL  0
 
In file included from D:/Xilinx/Vivado/2018.2/include/hls_fpo.h:166:0,
                 from D:/Xilinx/Vivado/2018.2/include/hls_half.h:58,
                 from D:/Xilinx/Vivado/2018.2/include/ap_int_sim.h:73,
                 from D:/Xilinx/Vivado/2018.2/include/ap_int.h:65,
                 from F:/OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_config.h:36,
                 from ../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_tb.cpp:32:
D:/Xilinx/Vivado/2018.2/include/floating_point_v7_0_bitacc_cmodel.h:135:0: note: this is the location of the previous definition
 #define __GMP_LIBGMP_DLL 1
 
In file included from D:/Xilinx/Vivado/2018.2/include/floating_point_v7_0_bitacc_cmodel.h:143:0,
                 from D:/Xilinx/Vivado/2018.2/include/hls_fpo.h:166,
                 from D:/Xilinx/Vivado/2018.2/include/hls_half.h:58,
                 from D:/Xilinx/Vivado/2018.2/include/ap_int_sim.h:73,
                 from D:/Xilinx/Vivado/2018.2/include/ap_int.h:65,
                 from F:/OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_config.h:36,
                 from ../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_accel.cpp:31:
D:/Xilinx/Vivado/2018.2/include/gmp.h:62:0: warning: "__GMP_LIBGMP_DLL" redefined
 #define __GMP_LIBGMP_DLL  0
 
In file included from D:/Xilinx/Vivado/2018.2/include/hls_fpo.h:166:0,
                 from D:/Xilinx/Vivado/2018.2/include/hls_half.h:58,
                 from D:/Xilinx/Vivado/2018.2/include/ap_int_sim.h:73,
                 from D:/Xilinx/Vivado/2018.2/include/ap_int.h:65,
                 from F:/OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_config.h:36,
                 from ../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_accel.cpp:31:
D:/Xilinx/Vivado/2018.2/include/floating_point_v7_0_bitacc_cmodel.h:135:0: note: this is the location of the previous definition
 #define __GMP_LIBGMP_DLL 1
 
obj/xf_colordetect_tb.o: In function `main':
F:\Vivadoproject\HLS\colordecete\solution1\csim\build/../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_tb.cpp:93: undefined reference to `cv::imread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
F:\Vivadoproject\HLS\colordecete\solution1\csim\build/../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_tb.cpp:184: undefined reference to `cv::imwrite(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
F:\Vivadoproject\HLS\colordecete\solution1\csim\build/../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_tb.cpp:185: undefined reference to `cv::imwrite(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
F:\Vivadoproject\HLS\colordecete\solution1\csim\build/../../../../../../OpenHW/xfopencv-master/examples/colordetect/xf_colordetect_tb.cpp:186: undefined reference to `cv::imwrite(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
collect2.exe: error: ld returned 1 exit status
make: *** [csim.exe] Error 1
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source F:/Vivadoproject/HLS/colordecete/solution1/csim.tcl"
    invoked from within
"hls::main F:/Vivadoproject/HLS/colordecete/solution1/csim.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C simulation.
`

bgouthamb, 2018-07-27:
@Yidagithub1103 
In Windows Vivado HLS 2018.2, there seems to be a problem with linking OpenCV libraries. Please use Vivado HLS 2017.4 with 2018.2 xfOpenCV code.

Yidagithub1103, 2018-07-27:
@bgouthamb 
Thanks for your reply!I will try later.
In your HLS_UseModel,you mentioned about "Use of appropriate compile-time options",I don't know what does "the time of compilation" mean.
Concretely speaking,I want to build your colordetect_accel in my PYNQ.I use webcam to get the video,and then do the color-detect.It involves the connection of PS and PL and building an IP to accelerate the function.So what should I do?

Yidagithub1103, 2018-07-27:
@bgouthamb 
I have tried 2018.2 xfOpenCV code in Vivado HLS 2017.4.
I meet same error like the first question.

bgouthamb, 2018-07-30:
@Yidagithub1103 
The problem is with the _-std=c++0x_ flag. Looks like some incompatibility with Windows Vivado HLS compiler.  Please do not include that flag during synthesis. We'll be updating the document accordingly.

tangjie77wd, 2018-08-16:
@Yidagithub1103 
Hello,i am a new one in the field of Sdsoc and HLS and i am confused about the choice Sdsoc or HLS , which one should i choose for image process platform . As you both said ,there are xfOpenCV function library in Sdsoc and hls function library in HLS . for instance, when i wanna get the sobel image for a source image,  xf::Sobel() or hls::Sobel()? Sdx  ISE or Vivado HLS, which one should i turn to?

tangjie77wd, 2018-08-16:
Hello,i am a new one in the field of Sdsoc and HLS and i am confused about the choice Sdx ISE or Vivado HLS , which one should i choose for image process platform . As you both said ,there are xfOpenCV function library in Sdx and hls function library in Vivado HLS . for instance, when i wanna get the sobel image for a source image, xf::Sobel() or hls::Sobel()? Sdx ISE or Vivado HLS, which one should i turn to?

tangjie77wd, 2018-08-16:
@Yidagithub1103 @bgouthamb  
Hello,i am a new one in the field of Sdsoc and HLS and i am confused about the choice Sdx ISE or Vivado HLS , which one should i choose for image process platform . As you both said ,there are xfOpenCV function library in Sdx and hls function library in Vivado HLS . for instance, when i wanna get the sobel image for a source image, xf::Sobel() or hls::Sobel()? Sdx ISE or Vivado HLS, which one should i turn to?

bgouthamb, 2018-08-17:
@tangjie77wd 
xf::Sobel() is the function to use. You can use the same xfOpenCV function in both SDx (to generate bitstream) and Vivado HLS (to perform C-Simulation, Synthesis etc..).

tangjie77wd, 2018-08-21:
@bgouthamb 
Thanks for your reply !
1) Do you mean i'd better take use of Sdx when i want bitstream by xfOpenCV and turn to Vivado HLS when i want simulation result or  synthesis IP file by xfOpenCV ?
2) As you know, Sdx ISE 2017.4 only support for 4 platforms( zc702, zc706, zcu102, zed ) . I added a custom platform yesterday , and now i wanna ceate xfOpenCV project on the custom platform,what should i do ? Do i need to transplant xfOpenCV to my custom platform or just need click 'Xilinx' ---'Sdx Libraray'----Xilinx 'xfOpenCV library '---'install'  in Sdx ISE 2017.4 ?

tangjie77wd, 2018-08-21:
@bgouthamb 
Thanks for your reply !

Do you mean i'd better take use of Sdx when i want bitstream by xfOpenCV and turn to Vivado HLS when i want simulation result or synthesis IP file by xfOpenCV ?
As you know, Sdx ISE 2017.4 only support for 4 platforms( zc702, zc706, zcu102, zed ) . I added a custom platform yesterday , and now i wanna ceate xfOpenCV project on the custom platform,what should i do ? Do i need to transplant xfOpenCV to my custom platform or just need click 'Xilinx' ---'Sdx Libraray'----Xilinx 'xfOpenCV library '---'install' in Sdx ISE 2017.4 ?
3) What is reVISION platform ? Does reVISION platform really means custom platform ?

tangjie77wd, 2018-08-21:
@bgouthamb 
Thanks for your reply !
1) Do you mean i'd better take use of Sdx when i want bitstream by xfOpenCV and turn to Vivado HLS when i want simulation result or synthesis IP file by xfOpenCV ?
2) As you know, Sdx ISE 2017.4 only support for 4 platforms( zc702, zc706, zcu102, zed ) . I added a custom platform yesterday , and now i wanna ceate xfOpenCV project on the custom platform,what should i do ? Do i need to transplant xfOpenCV to my custom platform or just need click 'Xilinx' ---'Sdx Libraray'----Xilinx 'xfOpenCV library '---'install' in Sdx ISE 2017.4 ?
3) What is reVISION platform ? Does reVISION platform really means custom platform ?

bgouthamb, 2018-08-21:
@tangjie77wd 

1. Yes.
2. You can separately download the xfOpenCV library from this GitHub page and follow the instructions 
    described under "Using the xfOpenCV Library on a non-reVISION Platform" section of [UG1233](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_2/ug1233-xilinx-opencv-user-guide.pdf).
3.  You can say reVISION platform as a custom platform in a way, with ready to build xfopencv file-i/o , 
     live-i/o samples and various other software modules as specified [here](https://github.com/Xilinx/Revision-Getting-Started-Guide/blob/master/overview.md#21-platform).

tangjie77wd, 2018-08-22:
@bgouthamb 
Thanks very much !
1) I am still comfused that reVISION platform actually means ZCU102 and ZCU104 board, am i right ?
2) Other platforms except for the 6 platforms( zc702, zc706, zcu102, zed,zcu102 ,zcu104 ) can be called non-reVISION Platform ,right ?
3) UG1233 is about  "Using the xfOpenCV Library on a reVISION Platform" and there is no section about "non-reVISION platform ".

bgouthamb, 2018-08-23:
@tangjie77wd 

1. There are ZCU104 and ZCU102 specific reVISION platforms targeting the respective boards.
2.  No. Any platform other than the ones listed [here](https://github.com/Xilinx/Revision-Getting-Started-Guide/blob/master/software-tools-system-requirements.md#32-software) (Design zip files) are non-reVISION platforms. So all these zc702, zc706, zcu102, zed, zcu102, zcu104 that come along with SDx are non-reVISION platforms too.
3. "Using the xfOpenCV Library on a non-reVISION Platform" section is available on page 13 of [UG1233](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_2/ug1233-xilinx-opencv-user-guide.pdf).

Please go through [reVISION Getting Started Guide](https://github.com/Xilinx/Revision-Getting-Started-Guide/blob/master/README.md) for complete details.

closed 2018-12-21 by bgouthamb