Xilinx/xfopencv #16: Example gaussianFilter fails simulation in HLS standalone

alionmgriffin, 2018-09-10:
System:
OS: Windows 10.0.16299
CPU: i7-7820HQ
RAM: 16.0 GB
HLS Version: 2017.4
XFOpenCV Version: Master branch cloned Sept. 10, 2018, 1629 UTC

Project:
Part: ZCU102 (xczu9eg-ffvb1156-2-i)
Period: 10
Uncertainty: (blank)
Top Level Function: gaussian_filter_accel
Test Bench CFlags: -D__XFCV_HLS_MODE__ -I../../../sw/xfopencv/include -std=c++0x
Source CFlags: -D__XFCV_HLS_MODE__ -IE:/sw/xfopencv/include -std=c++0x
Input arguments: E:/t/xfopencv/gaussianfilter/data/im0.jpg

Issue:
The 'guassianFilter' example fails simulation when run in Vivado HLS. Using the default configuration, FILTER_SIZE_3 and NO, the simulation reaches SIGSEGV in the following calls if not commented out:

xf_gaussian_filter.hpp:
Line 362: buf[bottom][col] = _src_mat.data[read_index++]
Line 413: _out_mat.data[write_index++] = (P0);
Line 471: buf[row_ind][col] = _src_mat.data[read_index++];
Line 530: _out_mat.data[write_index++] = (P0);

According to the data and array sizing, there should not be an issue with reading from and writing to the arrays. This leads me to believe that there is another location being effected by the indexing. If so, these calls may not be causing the segfault.

alionmgriffin, 2018-09-11:
Thanks @bgouthamb ,
I have not found any other examples that are failing on simulation. Some examples are failing on compilation, synthesis, or co-simulation. I am trying the examples on another Windows 10 machine as well as a Windows 7 and Linux machine. All the tests will be in 2017.4 HLS, due to the OpenCV linking issue in 2018.2, with a clone of the Master branch. Do you have a recommendation on which environment is best for HLS development?

bgouthamb, 2018-09-12:
@alionmgriffin 
Linux is recommended since Windows Vivado HLS is known to have issues during Synthesis, with  _-std=c++0x_ flag. Also, OpenCV linking issue doesn't exist in Linux for 2018.1/2018.2.

alionmgriffin, 2018-09-12:
@bgouthamb 
Tested in Linux, Windows 7, and another Windows 10. All were successful so, it is an issue with my machine. Have a good day.

closed 2018-09-12 by alionmgriffin