Xilinx/xfopencv #22: When I build xf_houghlines example，I meet this error!

swordwest, 2018-10-12:
===>The following messages were generated while processing C:/xf/houghlines/Release/_sds/p0/vivado\prj\prj.runs\impl_1 :
ERROR: [VPL 30-640] Place Check : This design requires more RAMB18 and RAMB36/FIFO cells than are available in the target device. This design requires 747 of such cell types but only 624 compatible sites are available in the target device. Please analyze your synthesis results and constraints to ensure the design is mapped to Xilinx primitives as expected. If so, please consider targeting a larger device.
ERROR: [VPL 30-99] Placer failed with error: 'IO Clock Placer stopped due to earlier errors. Implementation Feasibility check failed, Please see the previously displayed individual error or warning messages for more details.'
Please review all ERROR and WARNING messages during placement to understand the cause for failure.
ERROR: [VPL 17-69] Command failed: Placer could not place all instances
ERROR: [VPL 60-704] Integration error, problem implementing dynamic region, place_design ERROR
ERROR: [VPL 60-806] Failed to finish platform linker
ERROR: [SdsCompiler 83-5019] Exiting sds++ : Error when calling 'C:/Xilinx/SDx/2018.2/bin/vpl   --iprepo C:/xf/houghlines/Release/_sds/iprepo/repo  --iprepo C:/Xilinx/SDx/2018.2/data/ip/xilinx  --platform D:/download/fpga/zcu104-rv-ss-2018-2/zcu104_rv_ss/zcu104_rv_ss.xpfm  --temp_dir C:/xf/houghlines/Release/_sds/p0  --output_dir C:/xf/houghlines/Release/_sds/p0/vpl  --input_file C:/xf/houghlines/Release/_sds/p0/.xsd/top.bd.tcl  --target hw   --save_temps  --kernels w0_xf_HoughLines:adapter --webtalk_flag SDSoC  --remote_ip_cache C:/xf/ip_cache --xp \"param:compiler.deleteDefaultReportConfigs=false\" '
sds++ log file saved as C:/xf/houghlines/Release/_sds/reports/sds.log
ERROR: [SdsCompiler 83-5004] Build failed

make: *** [houghlines.elf] 错误 1

bgouthamb, 2018-10-12:
It essentially means that the resources available on the device aren't enough to implement the given function. For zcu104 device, Hough lines function can fit on it, up to a maximum image size configuration of 1920x1080.

swordwest, 2018-10-12:
My platform was  zcu104_rv_ss?Why is this problem？

bgouthamb, 2018-10-12:
Either you use the Base zcu104 platform or EV zcu104_rv_ss platform, the underlying FPGA is same (xczu7ev-ffvc1156-2).

swordwest, 2018-10-14:
But, How can I fix this error? Thanks for help!

bgouthamb, 2018-10-15:
Either you can reduce the image size or target your design to a larger device like zcu102.

closed 2018-12-21 by bgouthamb