Xilinx/xfopencv #24: Lauch HLS error in the SDx IDE

swordwest, 2018-10-15:
I meet this error ,when I lauch HLS in the SDx IDE.

[](url)
[image]

varunsanthaseelan, 2018-10-15:
@swordwest , could you confirm if the project was built? If yes, could you list out the sequence of steps for us to try and replicate the issue?

swordwest, 2018-10-16:
08:54:47 **** Incremental Build of configuration Release for project Fast ****
make pre-build main-build 
sdsoc_make_clean Release
' '
'Building file: ../src/xf_fast_accel.cpp'
'Invoking: SDS++ Compiler'
sds++ -DPLATFORM_ZCU104 -Wall -O3 -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/features -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/glib-2.0 -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/lib/glib-2.0/include -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include -I"../src" -ID:\download\fpga\zcu104-rv-ss-2018-2\zcu104_rv_ss/sw/a53_linux/a53_linux/inc/xfopencv -I"D:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/7.2.0" -I"D:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/7.2.0/aarch64-xilinx-linux" -I"D:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/7.2.0/backward" -I"C:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1" -I"C:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/aarch64-linux-gnu" -I"C:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/backward" -I"C:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include" -I"C:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include-fixed" -c -fmessage-length=0 -MT"src/xf_fast_accel.o" -Wno-overloaded-virtual -Wno-unused-label -Wno-strict-overflow -Wno-uninitialized -Wno-unused-function -Wno-unused-variable -Wno-unknown-attributes -Wno-unused-local-typedefs -Wno-sign-compare -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/6.2.1 -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/6.2.1/aarch64-xilinx-linux -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/6.2.1/backward -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/glib-2.0 -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/lib/glib-2.0/include -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include/c++/6.2.1 -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include/c++/6.2.1/aarch64-linux-gnu -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include/c++/6.2.1/backward -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/lib/gcc/aarch64-linux-gnu/6.2.1/include -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/lib/gcc/aarch64-linux-gnu/6.2.1/include-fixed -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include -Wno-overloaded-virtual -Wno-unused-label -Wno-strict-overflow -Wno-uninitialized -Wno-unused-function -Wno-unused-variable -Wno-unknown-attributes -Wno-unused-local-typedefs -Wno-sign-compare -mstrict-align -hls-target 1 -MMD -MP -MF"src/xf_fast_accel.d" -MT"src/xf_fast_accel.o" -o "src/xf_fast_accel.o" "../src/xf_fast_accel.cpp" -sds-hw "xf::fast<1,0,2160,3840,1>" xf_fast_accel.cpp  -files ../libs/xfopencv/include/features/xf_fast.hpp  -clkid 2 -sds-end -sds-sys-config a53_linux -sds-proc a53_linux -sds-pf "D:\download\fpga\zcu104-rv-ss-2018-2\zcu104_rv_ss"
Processing -sds-hw block for xf::fast<1,0,2160,3840,1>
Create data motion intermediate representation

C:\xf\Fast\Release>C:\Xilinx\SDx\2018.2\llvm-clang\win64\llvm\bin\clang.exe -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/features -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/glib-2.0 -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/lib/glib-2.0/include -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include -I../src -ID:\download\fpga\zcu104-rv-ss-2018-2\zcu104_rv_ss/sw/a53_linux/a53_linux/inc/xfopencv -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/7.2.0 -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/7.2.0/aarch64-xilinx-linux -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/7.2.0/backward -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1 -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/aarch64-linux-gnu -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/aarch64-linux-gnu/include/c++/7.2.1/backward -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include -IC:/Xilinx/SDK/2018.2/gnu/aarch64/nt/aarch64-linux/lib/gcc/aarch64-linux-gnu/7.2.1/include-fixed -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/6.2.1 -ID:\download\fpga\zcu104-rv-ss-2018-2\petalinux\sdk\sysroots\aarch64-xilinx-linux/usr/include/c++/6.2.1/aarch64-xilinx [... cut ...]
INFO: [VPL 60-841] Created output file: C:/xf/Fast/Release/_sds/p0/vpl/system.hdf
INFO: [VPL 60-841] Created output file: C:/xf/Fast/Release/_sds/p0/vpl/_new_clk_freq
Software tracing enabled
Compile hardware access API functions
Link application ELF file
SD card folder created C:/xf/Fast/Release/sd_card
All user specified timing constraints are met.
sds++ log file saved as C:/xf/Fast/Release/_sds/reports/sds.log

'Finished building target: Fast.elf'
' '

10:01:19 Build Finished (took 1h:6m:31s.956ms)

bgouthamb, 2019-06-18:
This feature is not supported with xfOpenCV functions, as Vivado HLS doesn't support a template as Top Function.

closed 2019-06-18 by bgouthamb