Xilinx/xfopencv #27: HLS_Use_Model/AXI_Sample missing #include statement

dsp-avt, 2019-01-11:
OS: Windows 10
Tool Version: 2018.3

The HLS_Use_Model/AXI_Sample is missing a #include "common/xf_axi.h" statement that causes C Simulation to fail.  Adding that #include statement to the bottom of the xf_header.h file fixes the issue.  

Here's the error before adding the #include statement:
[image]

Also, I had to increase the stack space in order to run the simulation.  I saw the following error which I tracked down to running out of memory:
[image]

I increased the stack space to 1GB by using the menu bar to navigating to Project > Project Settings > Simulation.
[image]

bgouthamb, 2019-01-16:
Hello @dsp-avt ,

Thanks for pointing out the missing #include. Corrected it.

Regarding the stack space issue, the __XFCV_HLS_MODEL__ flag makes the xf::Mat object to form array(s ) of fixed size, instead of pointer(s), to facilitate standalone HLS Synthesis and Co-Sim. These array(s) consume the stack memory heavily. Hence, increasing the stack size is necessary when working with larger image sizes (especially in Windows which has a very less default stack size of 1MB). This constraint will be addressed in the upcoming release.

bgouthamb, 2019-06-18:
Issue has been resolved in the latest release. Reopen if needed.

closed 2019-06-18 by bgouthamb