Xilinx/xfopencv #31: Mismatch between xf::Sobel documentation and implementation

andreasmuetzel-ametek, 2019-01-25:
On page 210 of UG1233 (https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug1233-xilinx-opencv-user-guide.pdf), the Sobel filter kernel for the x direction is given as 

<pre>-3 -10 -3
 0   0  0
 3  10  3
</pre>
However, this is the Scharr kernel in y direction and the code of xf::Sobel also uses the normal Sobel kernel:
<pre>-1 0 1
-2 0 2
-1 0 1</pre>
Am I right that this is simply an error in the documentation, and that it is not planned to make the 3x3 Sobel filter (silently) use the Scharr kernel?

I hope that this is the right place for this kind of "bug", sorry if it is not...

bgouthamb, 2019-01-25:
Hello @andreasmuetzel-ametek ,

Yes, it is indeed a documentation error.  Thanks for pointing it out! Will be correcting it in the next edition of the document.

andreasmuetzel-ametek, 2019-01-25:
Great, thank you!

closed 2019-01-25 by andreasmuetzel-ametek