Xilinx/xfopencv #33: Symbol 'Array2xfMat' could not be resolved

microlijun, 2019-01-29:
Hi,
I want to  computes the histogram of given input image and put the output into a Mat. 
But I got error: 'Symbol 'Array2xfMat' could not be resolved'    

       `unsigned int *g_hist;
	xf::Mat<XF_8UC1, HEIGHT, WIDTH, XF_NPPC1> &hist;
	xf::calcHist<XF_8UC1, HEIGHT, WIDTH, XF_NPPC1>(input_img,g_hist);
	xf::Array2xfMat< 8,XF_8UC1, HEIGHT, WIDTH, XF_NPPC1>(g_hist, hist);`

microlijun, 2019-01-31:
> 
> 
> `xf::Array2xfMat` is defined in _xf_utility.h_ file.
> 
> For this function to work, the memory allocation for the input pointer should be equal to the size of the input xf::Mat.

The output of xf::calcHist is an array of 256 elements, so 
xf::Mat<XF_8UC1, 256, 1, XF_NPPC1> &hist;
xf::calcHist<XF_8UC1, HEIGHT, WIDTH, XF_NPPC1>(input_img,g_hist);
xf::Array2xfMat< 8,XF_8UC1, 256, 1, XF_NPPC1>(g_hist, hist);

But still, "Invalid argumen"

bgouthamb, 2019-02-01:
You have specified the PTR_WIDTH as 8, but passing a 32-bit pointer. You need to call it as:
`xf::Array2xfMat< 32, XF_8UC1, 256, 1, XF_NPPC1>(g_hist, hist);`

closed 2019-02-01 by microlijun