Xilinx/xfopencv #34: get "Simulation failed: SIGSEGV" when I use tcl script to build AXI_Sample

yinedeng, 2019-01-30:
hi,
    when i use tcl script to build AXI_Samle, i get a /opt/sdsoc_2018.3/Vivado/2018.3/bin/vivado_hls script.tcl
   
    the building flow: 
    
    1、clone xfopencv repo and enter AXI_Samle directory
    2、do "/opt/sdsoc_2018.3/Vivado/2018.3/bin/vivado_hls script.tcl"

    finally, i get this error message:

--------------------------------------------------------------------------------------------
/opt/sdsoc_2018.3/Vivado/2018.3/include/opencv2/flann/flann.hpp:232:2: note: declared here
  Index_ {
  ^~~~~~
/opt/sdsoc_2018.3/Vivado/2018.3/include/opencv2/flann/flann.hpp:365:5: warning: ‘template<class T> class cv::flann::Index_’ is deprecated [-Wdeprecated-declarations]
 int Index_<T>::radiusSearch(const Mat& query, Mat& indices, Mat& dists, DistanceType radius, const ::cvflann::SearchParams& searchParams)
     ^~~~~~
/opt/sdsoc_2018.3/Vivado/2018.3/include/opencv2/flann/flann.hpp:232:2: note: declared here
  Index_ {
  ^~~~~~
@E Simulation failed: SIGSEGV.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source script.tcl"
    invoked from within
"hls::main script.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
INFO: [Common 17-206] Exiting vivado_hls at Wed Jan 30 13:48:33 2019...

bgouthamb, 2019-01-30:
Hello @yinedeng ,

You need to follow the [HLS use model guide](https://github.com/Xilinx/xfopencv/blob/master/HLS_Use_Model/HLS_UseModel_Usage_Doc.pdf) for building any xfOpenCV functions/applications for Vivado HLS.

yinedeng, 2019-01-31:
Hello @bgouthamb，
Thank you for your reply!
I had already read the guide. The guide say "using tlc script" is also one practicable to use in the chapter "Operating HLS Standalone mode". 

the script.tcl I used is from AXI_Samle and I had never changed it。

so, i can't execute the script on command line with "vivado_hls script.tcl" to build AXI_Samle ?

as far, i use script.tcl as a template, and change it to build the meanshifttracking example。it runs weill.

AXI_Samle is very important to me to demonstrate that I can use good axi stream interface to start my work。

chc0922, 2019-01-31:
It could be a stack size problem
refer this : 
https://github.com/Xilinx/xfopencv/issues/27

yinedeng, 2019-01-31:
@chc0922

Thank you very much! I have solved my problem.
Yes, It should be the stack size problem.
I used ubuntu 16.04  with 8k default stack size

It's succeeded to csim, csyn and co-sim when i set “ulimit -c unlimited ” before execute build script.

But I also found a problem makes me so confused。

I can't set "-Wl,--stack=SIZE" under vivado 2017.4 and "-z stack=SIZE" under vivado 2018.3 in GUI mode。

And，script mode also failed
csim_design -ldflags {-Wl,--stack,10485760}
cosim_design -ldflags {-Wl,--stack,10485760}

closed 2019-01-31 by yinedeng