Xilinx/xfopencv #37: xf_resize fails compiling on HLS

Q-point, 2019-02-06:
I tried to use a standalone compilation of resize, Gaussian, erode using HLS.

Only erode passes:

INFO: [SCHED 204-61] Option 'relax_ii_for_timing' is enabled, will increase II to preserve clock frequency constraints.
INFO: [HLS 200-10] Analyzing design file './xf_resize_accel.cpp' ...
ERROR: [HLS 200-70] Compilation errors found: In file included from ./xf_resize_accel.cpp:1:
In file included from ./xf_resize_accel.cpp:30:
In file included from ./xf_resize_config.h:37:
In file included from ../../include\imgproc/xf_resize.hpp:34:
In file included from ../../include\imgproc/xf_resize_headers.h:40:
In file included from ../../include\imgproc/xf_resize_nn_bilinear.hpp:38:
In file included from C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\iostream:39:
In file included from C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ostream:39:
In file included from C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ios:39:
In file included from C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception:151:
C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception_ptr.h:132:13: error: unknown type name 'type_info'
      const type_info*
            ^
In file included from ./xf_resize_accel.cpp:1:
In file included from ./xf_resize_accel.cpp:30:
In file included from ./xf_resize_config.h:37:
In file included from ../../include\imgproc/xf_resize.hpp:34:
In file included from ../../include\imgproc/xf_resize_headers.h:40:
In file included from ../../include\imgproc/xf_resize_nn_bilinear.hpp:38:
In file included from C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\iostream:39:
In file included from C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ostream:39:
In file included from C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\ios:39:
In file included from C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\exception:151:
C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:62:5: error: the parameter for this explicitly-defaulted copy constructor is const, but a member or base requires it to be non-const
    nested_exception(const nested_exception&) = default;
    ^
C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:64:23: error: the parameter for this explicitly-defaulted copy assignment operator is const, but a member or base requires it to be non-const
    nested_exception& operator=(const nested_exception&) = default;
                      ^
C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:77:28: error: exception specification in declaration does not match previous declaration
  inline nested_exception::~nested_exception() = default;
                           ^
C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:66:20: note: previous declaration is here
    inline virtual ~nested_exception();
                   ^
C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:122:61: error: redefinition of default argument
    __throw_with_nested(_Ex&& __ex, const nested_exception* = 0)
                                                            ^ ~
C:/Xilinx/Vivado/2018.3/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\nested_exception.h:110:56: note: previous definition is here
    __throw_with_nested(_Ex&&, const nested_exception* = 0)
                                                       ^ ~
In file included from ./xf_resize_accel.cpp:1:
In file included from ./xf_resize_accel.cpp:30:
In file included from ./xf_resize_config.h:34:
In file included from C:/Xilinx/Vivado/2018.3/common/technology/autopilot\ap_int.h:349:
In file included from C:/Xilinx/Vivado/2018.3/common/technology/autopilot\ap_fixed.h:55:
C:/Xilinx/Vivado/2018.3/common/technology/autopilot\ap_fixed_base.h:837:13: warning: shift count is negative [-Wshift-count-negative]
      ret.V <<= (_AP_I - _AP_W);
            ^   ~~~~~~~~~~~~~~~
C:/Xilinx/Vivado/2018.3/common/technology/autopilot\ap_int_base.h:361:18: note: in instantiation of member function 'ap_fixed_base<50, 21, true, 5, 3, 0>::to_ap_int_base' requested here
    Base::V = op.to_ap_int_base().V;
                 ^
C:/Xilinx/Vivado/2018.3/common/technology/autopilot\ap_int.h:231:9: note: in instantiation of function template specialization 'ap_int_base<17, false>::ap_int_base<50, 21, true, 5, 3, 0>' requested here
      : Base((ap_fixed_base<_AP_W2, _AP_I2, true, _AP_Q2, _AP_O2, _AP_N2>)op) {}
        ^
../../include\imgproc/xf_resize_nn_bilinear.hpp:173:17: note: in instantiation of function template specialization 'ap_uint<17>::ap_uint<50, 21, 5, 3, 0>' requested here
   nextYScale = indexy_pre+scaley;
                ^
../../include\imgproc/xf_resize_nn_bilinear.hpp:312:5: note: in instantiation of function template specialization 'computeInterpolation<2, 42, 20, 17, 32, 3, 12, 2, 1>' requested here
    computeInterpolation<INTERPOLATION_TYPE, COMP_INDEX_WIDTH, COMP_INDEX_INT, INDEX_INT, SCALE_WIDTH, SCALE_INT, WEIGHT_WIDTH, WEIGHT_INT, NPPC>(imgInput.rows, imgInput.cols, j<<(xfNPixelsPerCycle<NPPC>::datashift), output_rows_count, scalex, scaleXParallel, scaley, indexx, indexy, nextYScale, WeightX, WeightY, indexx_pre_comp, indexy_pre_comp);
    ^
../../include\imgproc/xf_resize.hpp:86:3: note: in instantiation of function template specialization 'resizeNNBilinear<0, 2160, 3840, 1, 1080, 1920, 2, 2>' requested here
  resizeNNBilinear<TYPE, SRC_ROWS, SRC_COLS, NPC, DST_ROWS, DST_COLS, INTERPOLATION_TYPE, MAX_DOWN_SCALE>(_src,_dst);
  ^
[... 66 lines cut ...]
C:/Xilinx/Vivado/2018.3/common/technology/autopilot\ap_int.h:231:9: note: in instantiation of function template specialization 'ap_int_base<8, false>::ap_int_base<32, 22, true, 5, 3, 0>' requested here
      : Base((ap_fixed_base<_AP_W2, _AP_I2, true, _AP_Q2, _AP_O2, _AP_N2>)op) {}
        ^
../../include\imgproc/xf_resize_nn_bilinear.hpp:65:12: note: in instantiation of function template specialization 'ap_uint<8>::ap_uint<32, 22, 5, 3, 0>' requested here
   pixel = (typename DataType<DEPTH,NPPC>::cname)((ap_fixed<32,22>)(P1 + P2 + P3 + P4));
           ^
../../include\imgproc/xf_resize_nn_bilinear.hpp:115:5: note: in instantiation of function template specialization 'interpolatePixel<0, 2, 1>' requested here
    interpolatePixel<DEPTH, INTERPOLATION_TYPE, NPPC>(unpackX1temp[indexx[i]-initIndex], unpackX2temp[indexx[i]-initIndex], unpackX1temp[indexx[i]-initIndex+1], unpackX2temp[indexx[i]-initIndex+1], Wx[i], Wy, currentoutput);
    ^
../../include\imgproc/xf_resize_nn_bilinear.hpp:410:7: note: in instantiation of function template specialization 'computeOutputPixel<0, 2, 1, 17, 2>' requested here
      computeOutputPixel<SRC_TYPE,INTERPOLATION_TYPE,NPPC,INDEX_INT,BUFFER_WORDS>(P0Buf,P1Buf,((indexx[0]>>xfNPixelsPerCycle<NPPC>::datashift)<<xfNPixelsPerCycle<NPPC>::datashift),indexx,WeightX,WeightY,temp_store_output);
      ^
../../include\imgproc/xf_resize.hpp:86:3: note: in instantiation of function template specialization 'resizeNNBilinear<0, 2160, 3840, 1, 1080, 1920, 2, 2>' requested here
  resizeNNBilinear<TYPE, SRC_ROWS, SRC_COLS, NPC, DST_ROWS, DST_COLS, INTERPOLATION_TYPE, MAX_DOWN_SCALE>(_src,_dst);
  ^
./xf_resize_accel.cpp:34:2: note: in instantiation of function template specialization 'xf::resize<2, 0, 2160, 3840, 1080, 1920, 1, 2>' requested here
 xf::resize <2, XF_8UC1, 2160, 3840, 1080, 1920, XF_NPPC1, 2> (_src, _dst);
 ^
4 warnings and 5 errors generated.
Failed during preprocessing.
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

bgouthamb, 2019-02-07:
@Q-point ,

Do not use the flag "-std=c++0x" for Synthesis, when using Vivado HLS in Windows OS. It is mentioned as a note in [HLS use model guide](https://github.com/Xilinx/xfopencv/blob/master/HLS_Use_Model/HLS_UseModel_Usage_Doc.pdf).

Q-point, 2019-02-07:
@bgouthamb

OK, thanks. Synthesis passes when I remove the flag "-std=c++0x".  Curiously enough the same flag works with the dilation core with no problem on Windows 10.

closed 2019-02-07 by Q-point

natu4u, 2020-06-22:
> @Q-point ,
> 
> Do not use the flag "-std=c++0x" for Synthesis, when using Vivado HLS in Windows OS. It is mentioned as a note in [HLS use model guide](https://github.com/Xilinx/xfopencv/blob/master/HLS_Use_Model/HLS_UseModel_Usage_Doc.pdf).

Hi bgaouthamb, I 'm facing the same issue. But on removing the "-std=c++0x" although the original errors are resolved, I'm facing a "C preprocesor failed" error as shown below : - 
______________________________________________________________________________________________________
In file included from C:/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\cstdint:34:
C:/Xilinx/Vivado/2019.2/win64/tools/clang/bin\..\lib\clang\3.1/../../../include/c++/4.5.2\bits/c++0x_warning.h:31:2: **error: This file requires compiler and library support for the upcoming ISO C++ standard, C++0x. This support is currently experimental, and must be enabled with the -std=c++0x or -std=gnu++0x compiler options.**
#error This file requires compiler and library support for the upcoming \
 ^
3 warnings and 1 error generated.
C preprocessor failed.
    while executing
"source C:/Xilinx/work/DiRAC_projects/DiRAC_master/src/DiRAC_trial2/solution1/csynth.tcl"
    invoked from within
"hls::main C:/Xilinx/work/DiRAC_projects/DiRAC_master/src/DiRAC_trial2/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"
Finished C synthesis.
______________________________________________________________________________________________

Any idea as to what might be the reason ?Thanks, Nitin