Xilinx/xfopencv #41: xf_warpperspective (or warpaffine) do not synthesize

ps-george, 2019-03-20:
```
void ip_accel(ap_uint<INPUT_PTR_WIDTH> * in_img, ap_uint<OUTPUT_PTR_WIDTH> * out_img, float H[9], int rows, int cols){
        const int pROWS = XF_HEIGHT;
        const int pCOLS = XF_WIDTH;
        const int pNPC1 = XF_NPPC8;
        const int pDepth = pROWS * pCOLS;

#pragma HLS INTERFACE m_axi port=in_img depth=pDepth offset=direct
#pragma HLS INTERFACE m_axi port=out_img depth=pDepth offset=direct

        xf::Mat<XF_8UC1, XF_HEIGHT, XF_WIDTH, XF_NPPC8> in_mat(rows, cols);
        xf::Mat<XF_8UC1, XF_HEIGHT, XF_WIDTH, XF_NPPC8> out_mat(rows, cols);

        xf::Array2xfMat<INPUT_PTR_WIDTH, XF_8UC1, XF_HEIGHT, XF_WIDTH, XF_NPPC8>(in_img, in_mat);
        xf::warpPerspective<XF_INTERPOLATION_BILINEAR, XF_8UC1, XF_HEIGHT, XF_WIDTH, XF_NPPC8>(in_mat, out_mat, H);
        xf::xfMat2Array<OUTPUT_PTR_WIDTH, XF_8UC1, XF_HEIGHT, XF_WIDTH, XF_NPPC8>(out_mat, out_img);
}
```

`ERROR: [SYNCHK 200-22] ../xfopencv/include/common/xf_utility.h:370: memory copy is not supported unless used on bus interface possible cause(s): non-static/non-constant local array with initialization).`

Due to the calls to `xFCopyBlockMemoryIn1`  and `xFCopyBlockMemoryOut1` in their implementation, which call `memcpy`.

I can call memcpy from top level with no issues, e.g. the following synthesizes fine:
```
void memcpy_test(unsigned long long int *ddr){
        const int dsize = DATA_SIZE;
        int i;
        unsigned long long int data[DATA_SIZE];

#pragma HLS INTERFACE m_axi port=ddr depth=dsize offset=direct
        for (i=0;i<DATA_SIZE;i++){
                data[i] = (unsigned long long int)i;
        }
        memcpy(ddr, data, DATA_SIZE*sizeof(unsigned long long int));
}
```

But the issue is I can't (and shouldn't be) defining bus/m_axi interfaces for functions defined in xfopencv source.

What am I missing here?

bgouthamb, 2019-03-25:
The statement in Page no: 119 of [UG902 ](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug902-vivado-high-level-synthesis.pdf)says, "The C memcpy function is only supported for synthesis when used to transfer data to or from a top-level function argument specified with an AXI4 master interface"

In your case,  the memcpy in _xf::warpperspective_  is not happening on a top-level function argument,. You are doing a deep copy from the top-level argument(in_img) into another pointer location (in_mat.data) in the _Array2xfMat_ function and passing that to _xf::warpperspective_. Hence the error.

ps-george, 2019-03-27:
This solves this issue. I can now generate an IP block.

However, generated for my target board, and creating a minimal design (ZYNQ -> AXI Interconnect -> HLS Warp Perspective) requires more DSP blocks than are available.

closed 2019-03-27 by ps-george