Xilinx/xfopencv #45: How to build SDSoC application with xfOpencv to zedboard?

microlijun, 2019-03-31:
The tutorial build an SDSoC application project using that hardware accelerated xfOpenCV functions running on the ZCU102 reVISION platform, but I only have a zedboard,7020. 
How to build SDSoC application with xfOpencv to zedboard? Is there any instruction?

bgouthamb, 2019-04-01:
@microlijun Refer to the "Using the xfOpenCV Library on a non-reVISION
Platform" section of [UG1233](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug1233-xilinx-opencv-user-guide.pdf)

microlijun, 2019-04-05:
Hi @bgouthamb ,
I chose zed as hardware as hardware platform, cross complied opencv for arm-linux and added it to Include Paths. But I still got errors. Log file is attached below. 
Many thanks.
[sds_xf_accumulate_image_accel.log](https://github.com/Xilinx/xfopencv/files/3049748/sds_xf_accumulate_image_accel.log)

bgouthamb, 2019-04-08:
@microlijun ,
The flag _-mstrict-align_  is for AArch64. zed board has AArch32. Hence that flag should be removed from Makefile. (you may use _-mno-unaligned-access_, but it is not mandatory)

bgouthamb, 2019-06-18:
Reopen if issue not resolved.

closed 2019-06-18 by bgouthamb

quantom3230, 2020-12-01:
Hi @bgouthamb ,
can you help why I could not build in non revision platform in zed borad while I don't have this board on windows OS?
I dont know how and where I can get folder of opencv_libs properly. actually I have tested many opencv libs, but I couldn't build it.
sdx 2018.3 
OS: windows 10
I appreciate to help me
this one of my errors:

[log.txt](https://github.com/Xilinx/xfopencv/files/5625174/log.txt)