Xilinx/xfopencv #46: Elapsed time in xf_header.h

zjru, 2019-04-11:
Hi,

I have questions about the time count for the hardware accelerators in the xfOpenCV library. 

1. I tested the StereoBM and SGBM on zcu102 board but do not understand how the elapsed time is calculated. In "xf_headers.h", there is a line shown as "elapsed time ..." and a definition of TIME_STAMP. In the main function (tb.cpp), the header file is included but it seems that TIME_STAMP is not invoked. And the main function uses another API (perf_counter) to count the computation time of the hardware accelerator. In this case, why does the terminator still print out the "elapsed time" as shown in the xf_headers.h? 

2. Does the elapsed time printed mean the CPU clock cycles?

3. There is some difference between the "elapsed time" shown in the terminator and the estimation time from the "Estimation Performance Report". The estimation time from the report is the number of clock cycles (CPU time) estimated by sdx. Can we rely on this estimation value for implementation or optimization? Since it takes a much longer time to run the whole flow and test the actual time on board.

Thanks!

Best regards,
Jieru

zjru, 2019-04-11:
Also, why are the elapsed time printed in the terminator and the estimated computation time from the report different?

Thanks,
Jieru

bgouthamb, 2019-04-12:
@zjru 

1. Latency is not calculated with TIME_STAMP function. A more accurate _perf_counter_ object is used instead, which is defined in include/common/xf_sw_utils.h
2. Yes. 
3. As the name itself implies, the "Estimate Performance Report" gives an estimate of the latency/resource numbers generated based on the tripcount values/array sizes defined in the function([worst-case latency](https://github.com/Xilinx/SDSoC-Tutorials/blob/master/getting-started-tutorial/images/epe1526670476717.png)), whereas the elapsed time displayed when running on hardware is the exact latency for the given input image size.

zjru, 2019-04-12:
@bgouthamb Thanks for your reply!

The elapsed time seems not to include the communication time for data transfer. When computing some metric like frame per second (FPS),  do you use the elapsed time to compute it? 

Thanks!

bgouthamb, 2019-04-15:
@zjru 

The communication time for data transfer between DDR to FPGA is included in the elapsed time. 
Yes, FPS is calculated using this elapsed time.

zjru, 2019-04-16:
@bgouthamb Thanks!

closed 2019-04-16 by zjru