Xilinx/xfopencv #47: Harris Corner Detection Behavior

fmascarich, 2019-05-03:
I'm running the Harris corner detector example, and I've gotten some fairly strange results. 
This is my input image:
[image]
Here is the output_hls.png:
[image]
Here is the output_ocv.png:
[image]

bgouthamb, 2019-05-03:
@fmascarich 
There will be a minimal difference in the output images of xfOpenCV function (output_hls.png) and the reference function(output_ocv.png). 
Since xfOpenCV works on fixed point format and the reference function uses floating point, there will be a little variation in the computations, that causes the results to differ.

fmascarich, 2019-05-03:
That makes sense, however I intended to point out the differences when compared to the results of the official opencv implementation. Here's the result using the official opencv function:

[image]

output_hls finds corners on the border of the image, and output_ocv is missing a significant number of corners in a cross pattern on the central axis, I'm just wondering if there's something I'm not configuring correctly or if there's buggy behavior in the xfopencv implementation.

bgouthamb, 2019-05-08:
@fmascarich 
OpenCV's [cornerHarris](https://docs.opencv.org/2.4/modules/imgproc/doc/feature_detection.html?highlight=cornerharris#cornerharris) implementation returns Harris detector responses as output. To get the corners out of these scores, few post-processing functions has to be applied. The functions to be used for post-processing is totally dependent on the target application/image.

In xfOpenCV, threshold and max-suppression functions are the post-processing functions, whereas [OpenCV's example](https://docs.opencv.org/3.4.2/d4/d7d/tutorial_harris_detector.html) applies a different set of functions and later a threshold function with a threshold value suitable for that image, to get corners from Harris Scores.

If we apply the same post-processing functions as OpenCV, instead of max-suppression, to xfOpenCV cornerHarris, their output's both match.

Also, xfOpenCV's implementation only supports BORDER_CONSTANT. Hence, corners are being detected on the image borders.

Currently, in xfOpenCV, the post-processing functions (thresholding and max-suppression) are integrated into the cornerHarris function itself. These functions will be separated from cornerHarris in a future release.