Xilinx/xfopencv #50: Canny EdgeTracing error

estibalitz, 2019-05-15:
I am trying to implement the Canny edge detection in SDx 2017.4 using the release of 2017.4.
I imported the files in the folder Canny to do so, and I tried to solve some problems related to data type. But after all, what I am facing is that the EdgeTracing <XF_2UC1,XF_8UC1,HEIGHT, WIDTH, XF_NPPC32,XF_NPPC8>(_dst1,_dst2) is no implemened because it doesn match. 

I changed it into --> xf::xfEdgeTracing<XF_2UC1,XF_8UC1,HEIGHT, WIDTH, XF_NPPC32,XF_NPPC8>(_dst1,_dst2); because this is how is defined in xf_edge_tracing.hpp
xfEdgeTracing((unsigned long long *)_dst.data,(unsigned long long *)_src.data,_dst.rows,_dst.cols);

But I am getting the following error:

ERROR: [SdsCompiler 83-5005] clang exited with non-zero code processing /home/aranbarr/Documents/Algoritmos/003/reVISION-Zybo-Z7-20-2017.4-3/003/src/xf_canny_accel.cpp
ERROR: [SdsCompiler 83-5004] Build failed

src/subdir.mk:21: recipe for target 'src/xf_canny_accel.o' failed
make: *** [src/xf_canny_accel.o] Error 1

bgouthamb, 2019-05-16:
@estibalitz 

What steps did you follow to build the Canny function? 
Are you following the [UG1233](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2017_4/ug1233-xilinx-opencv-user-guide.pdf) document? 
Have you modified anything in any of the xfOpenCV files? 
Did you try building the Canny example as is?

estibalitz, 2019-05-16:
Yes, I am using the UG1233, I haven modifed anything in the xfOpenCV file
and I did try to run the example as it is...
but it gets stuck
[image: image.png]

Hau idatzi du Goutham Borra (notifications@github.com) erabiltzaileak (2019
mai. 16, og. (11:50)):

estibalitz, 2019-05-16:
I have just created the project as I was going to create a bilateral filter
demo, and in the src folder I replaced all the bilateral filters file for
the canny filter files.

As I was getting problems, I tried to run it in the CPU not in the HW, and
although this, it does not work

Hau idatzi du Maria Estibaliz Arambarri Cincunegui (
maria.arambarri@alumni.mondragon.edu) erabiltzaileak (2019 mai. 16, og.
(14:37)):

bgouthamb, 2019-05-17:
@estibalitz 

Have you followed the steps in "Using the xfOpenCV Library on a non-reVISION Platform" section or "Using reVISION Samples on the reVISION Platform" section of UG1233?
Are you getting similar issues with other functions too?
Can you post the complete log file(s)?

estibalitz, 2019-05-28:
I am using it in a reVision platform and I am not getting similar outputs with other functions.

estibalitz, 2019-05-29:
the errors are the following ones:

[image]

[image]

But I don't understand why there is a mismatch between the function definition and the actual original implementation.

bgouthamb, 2019-05-31:
@estibalitz 
From the log, it appears that you are using the 2018.3 version of xfOpenCV functions with the 2017.4 platform, in 2017.4 SDSoC. You need to use the same version of the tool, platform and the code base for your build. Also, it is recommended to create a new project every time you build a new function.