Xilinx/xfopencv #51: Can't find RGB2GRAY function in xfopencv

HungSoma, 2019-05-24:
Hi,
I can't find rgb2gray and gray2rgb in xfopencv when add xf_cvt_color.hpp into Vivado HLS , it only rgba2iyuv, rgba2nv12, rgba2nv21,.. 
And I can't use hls::cvtcolor with xf::Mat, pls help me! 
Regard,
Hung

bgouthamb, 2019-05-27:
Hello @HungSoma,

rgb2gray and gray2rgb configurations will be available in the next release of xfOpenCV.

For now, as a workaround, you may use the converter functions from [xf::Mat to AXIVideo](https://github.com/Xilinx/xfopencv/blob/d619711903c103ef61c91ac35b93211260b3ba3a/include/common/xf_infra.h#L173) and [AXIVideo to hls Mat](https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842537/HLS+AXIvideo2Mat) and then call the hls::cvtcolor.

HungSoma, 2019-05-28:
Thank you, I hope to see it  soon!

bgouthamb, 2019-06-18:
_rgb2gray_ and _gray2rgb_ configurations of cvtColor function are now available in 2019.1 release.

closed 2019-06-18 by bgouthamb

HungSoma, 2019-06-20:
Thanks all!