Xilinx/xfopencv #52: ERROR: [SYNCHK 200-71] function 'lround' has no function body.

3togo, 2019-06-03:
Any hints to fix this error?

ERROR: [SYNCHK 200-71] ../xfopencv/xfopencv-2018.3_release/include/common/xf_structs.h:503: function 'lround' has no function body.

bgouthamb, 2019-06-03:
@3togo 
'lround' is a standard C++ function, which cannot be synthesized. The _xf::Mat_ member function _convertTo_ is meant to be used only on the host side. The synthesizable _convertTo_ function is available at _include/imgproc/xf_convert_bitdepth.hpp_

3togo, 2019-06-03:
do you mean that xf::Mat has two versions? One is for host side and the other is for the synthetic side.

3togo, 2019-06-03:
How to do something similar to below
xf::convertTo<XF_16UC1, XF_8UC1, src_rows, src_cols, XF_NPPC1> (dst16, dst, XF_CONVERT_16U_TO_8U, 0);

that can be synthesized?

bgouthamb, 2019-06-03:
xf::convertTo<XF_16UC1, XF_8UC1, src_rows, src_cols, XF_NPPC1> (dst16, dst, XF_CONVERT_16U_TO_8U, 0);

This indeed is the synthesizable function. Are you getting the issue with 'lround' when synthesizing the above function? What is the SDx tool version you use?

3togo, 2019-06-03:
I am using it in bare HLS. It is not synthesizable using Vivado 2019.1 and 2018.3

bgouthamb, 2019-06-04:
@3togo 
Can you post the log file, along with the source file where you are calling this function?

3togo, 2019-06-24:
better focus on 2019.1 version instead

closed 2019-06-24 by 3togo