Xilinx/xfopencv #53: xfopencv triggers the use of wrong version of clang(3.1)  when using in barebone HLS

3togo, 2019-06-04:
Whenever, I want to use xfopencv to synthesis code, it will trigger the following errors.  

Strangely enough this bug happens in Vivaldo HLS 2018.3 and 2019.1 but not in 2018.2

ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_storeups': /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/xmmintrin.h:630
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_movntps'; did you mean '__builtin_ia32_movntss'?: /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/xmmintrin.h:678
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_cvtps2pd': /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/emmintrin.h:380
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_cvtdq2pd'; did you mean '__builtin_ia32_cvtdq2ps'?: /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/emmintrin.h:386
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_storeupd': /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/emmintrin.h:586
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_pavgb128': /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/emmintrin.h:671
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_pavgw128': /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/emmintrin.h:677
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_storedqu': /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/emmintrin.h:1177
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_storelv4si'; did you mean '__builtin_ia32_storelps'?: /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/emmintrin.h:1189
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_movntpd'; did you mean '__builtin_ia32_movntsd'?: /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/emmintrin.h:1195
ERROR: [HLS 214-124] use of undeclared identifier '__builtin_ia32_movntdq'; did you mean '__builtin_ia32_movntq'?: /tools/Xilinx/Vivado/2019.1/lnx64/tools/clang/lib/clang/3.1/include/emmintrin.h:1201

bgouthamb, 2019-06-06:
@3togo 
Can you post the files that you are trying to synthesize, along with the complete log file?

3togo, 2019-06-09:
I assume that opencv and xfopencv are installed in /opt
/opt/opencv2/include
/opt/xfopencv/include
[dilation.zip](https://github.com/Xilinx/xfopencv/files/3269850/dilation.zip)

bgouthamb, 2019-06-10:
@3togo 
You are not supposed to include non-synthesizable files like 
```
opencv/cv.h
opencv2/highgui.hpp
common/xf_sw_utils.h
common/xf_axi.h
```
in the file targeted for synthesis. 

or

The files/code that is not intended for synthesis should be masked with the pre-defined macro `__SYNTHESIS__`

Example considering your _axi_config.h_:

```
#ifndef `__SYNTHESIS__`
#include  "common/xf_sw_utils.h"
#include  "common/xf_axi.h"
#include "opencv2/opencv.hpp"
#include  "opencv2/imgproc.hpp"
#include "opencv/cv.h"
#include "opencv2/core/core.hpp"
#include  "opencv2/highgui.hpp"
#endif

#include  "hls_stream.h"
#include "common/xf_common.h"
#include "common/xf_utility.h"
#include  "common/xf_infra.h"
#include "common/xf_axi_sdata.h"
typedef hls::stream< ap_axiu<24,1,1,1> > AXI_STREAM;
/////////////////////////////////////////////////////////////

void dilation_accel(AXI_STREAM& img_data_l, AXI_STREAM& img_data_r, AXI_STREAM& img_data_d);
```

Go through [UG902 ](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2018_3/ug902-vivado-high-level-synthesis.pdf) to understand HLS Synthesis.

3togo, 2019-06-10:
bgouthamb,
 
Many thanks . Problem Solved

closed 2019-06-10 by 3togo