Xilinx/xfopencv #55: HLS AXI standalone fails sythesizing under 2018.3

Q-point, 2019-06-17:
**I built the project issuing vivado_hls -f scrip.tcl.
Simulation passes but synthesizing fails.**

INFO: [HLS 200-10] Checking synthesizability ...
INFO: [XFORM 203-602] Inlining function 'xf::Mat<0, 1080, 1920, 1>::write' into 'xf::AXIvideo2xfMat<8, 0, 1080, 1920, 1>' (../../include\common/xf_infra.h:51->../../include\common/xf_infra.h:61->../../include\common/xf_infra.h:150) automatically.
Stack dump:
0.	Running pass 'User-controlled function inlining' on module 'C:/Users/aa/Downloads/ZYNQProject/xfopencv-master/HLS_Use_Model/Standalone_HLS_AXIS_Example/dilation_project/solution1/.autopilot/db/a.g.1.bc'.
0x00007FFB1C46DCC8 (0x0000000000000000 0x000001D6B50BC370 0x0000006004BF07A0 0x000001D6B3E95D00), ?main@Syn@@YAHHPEAPEAD@Z() + 0x30DD78 bytes(s)
0x00007FFB1C51A006 (0x000001D6B41DA562 0x000001D6B5C88240 0x0000000000000000 0x000000000000000F), ?main@Syn@@YAHHPEAPEAD@Z() + 0x3BA0B6 bytes(s)
0x00007FFB1CB3B5B6 (0x0000000000000001 0x000001D6B41DA562 0x0000000000000001 0x0000000000000027), ?save_object_ptr@?$pointer_oserializer@Vxml_oarchive@archive@boost@@VTransition@DBFsm@fsmd@@@detail@archive@boost@@EEBAXAEAVbasic_oarchive@234@PEBX@Z() + 0x390B36 bytes(s)
0x00007FFB1CB32C6C (0x000001D6B41DA562 0x000001D6B41DA562 0x000001D6B41DA562 0x0000006004BF1068), ?save_object_ptr@?$pointer_oserializer@Vxml_oarchive@archive@boost@@VTransition@DBFsm@fsmd@@@detail@archive@boost@@EEBAXAEAVbasic_oarchive@234@PEBX@Z() + 0x3881EC bytes(s)
0x00007FFB1CB3BE52 (0x0000006000000000 0x000001D6B5C89C80 0x0000000000000000 0x000000000000001C), ?save_object_ptr@?$pointer_oserializer@Vxml_oarchive@archive@boost@@VTransition@DBFsm@fsmd@@@detail@archive@boost@@EEBAXAEAVbasic_oarchive@234@PEBX@Z() + 0x3913D2 bytes(s)
0x00007FFB1CB3CA80 (0x0000000000000000 0x000001D6B3DD72F0 0x0000000000000000 0x0000000000000000), ?save_object_ptr@?$pointer_oserializer@Vxml_oarchive@archive@boost@@VTransition@DBFsm@fsmd@@@detail@archive@boost@@EEBAXAEAVbasic_oarchive@234@PEBX@Z() + 0x392000 bytes(s)
0x00007FFB1C450DB9 (0x0000000000000000 0x000001D6B52D59B0 0x0000000000000000 0x0000BECB831C8779), ?main@Syn@@YAHHPEAPEAD@Z() + 0x2F0E69 bytes(s)
0x00007FFB1C44FCA0 (0x0000000000000000 0x0000006004BF1CD0 0x000001D6B52D59B0 0x0000000000000000), ?main@Syn@@YAHHPEAPEAD@Z() + 0x2EFD50 bytes(s)
0x00007FFB1C10BB5D (0x0000000000000000 0x000001D6B3C11B80 0x0000006004BF1CD0 0x00007FFB1D57A210)
0x00007FFB31799FE3 (0x0000000000000001 0x0000000000000000 0x0000000000000000 0x000001D6AF6A8E70), ?addOwnedOption@TclCommand@xpcl@@QEAAXPEAVOption@2@@Z() + 0x3B3 bytes(s)
0x00007FFB3179F702 (0x000001D6B3DB8DDD 0x000001D6B3DB8DDB 0x000001D6B3940BF0 0x0000000071510E7B), ?setResultObj@TclCommand@xpcl@@QEAAXPEAUTcl_Obj@@@Z() + 0x62 bytes(s)
0x00000000714611FC (0x000001D600000000 0x000001D600000054 0x000001D6AF6B0A28 0xFFFFFFFFFFFFFFFF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x00000000714A6238 (0x000001D6AF6A8E70 0x000001D6B3D90F60 0x0000000000000000 0x000001D600000000), Tcl_ExprObj() + 0x1868 bytes(s)
0x00000000714A4FEC (0x000001D6AF31CFC0 0x000001D6AF6A8E70 0x0000000000000001 0x000001D6AF6A8E70), Tcl_ExprObj() + 0x61C bytes(s)
0x0000000071462E8D (0x000001D6B1F72210 0x0000000000000003 0x000001D6AF6A8E70 0x000001D6B0FE25E0), TclEvalObjEx() + 0x3AD bytes(s)
0x0000000071462ACD (0x000001D6AF6B09D0 0x000001D6AF25C3B0 0x000001D6AF31CFC0 0x00000000714F4790), Tcl_EvalObjEx() + 0x1D bytes(s)
0x00007FFB23B0075B (0x000001D6AF6A8E70 0x000001D6AF6A8E70 0x000001D6AF6B09B8 0x000000007152AC7D)
0x00000000714611FC (0x0000000000000000 0x0000000000000003 0x000001D6AF6B09B8 0xFFFFFFFFFFFFFFFF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x00000000714A6238 (0x000001D6AF6A8E70 0x000001D6B29B09A0 0x000001D6AF6B07E0 0x0000000000000000), Tcl_ExprObj() + 0x1868 bytes(s)
0x00000000714F12F4 (0x0000000000000000 0x000001D6AF6A8E70 0x000001D6AF6B07E0 0x000000007152AC7D), TclObjInterpProcCore() + 0x74 bytes(s)
0x00000000714611FC (0x0000000000000000 0x0000000000000003 0x000001D6AF6B07E0 0xFFFFFFFFFFFFFFFF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x00000000714A6238 (0x000001D6AF6A8E70 0x000001D6B3D9B090 0x0000000000000000 0x000001D600000000), Tcl_ExprObj() + 0x1868 bytes(s)
0x00000000714A4FEC (0x000001D6AF31CFC0 0x000001D6AF6A8E70 0x0000000000000001 0x000001D6AF6A8E70), Tcl_ExprObj() + 0x61C bytes(s)
0x0000000071462E8D (0x000001D6B3CB2D00 0x0000000000000003 0x000001D6AF6A8E70 0x000001D6B0FE25E0), TclEvalObjEx() + 0x3AD bytes(s)
0x0000000071462ACD (0x000001D6AF6B0780 0x000001D6AF25C3B0 0x000001D6AF31CFC0 0x00000000714F4790), Tcl_EvalObjEx() + 0x1D bytes(s)
0x00007FFB23B0075B (0x000001D6AF6A8E70 0x000001D6AF6A8E70 0x000001D6AF6B0768 0x000000007152AC7D)
0x00000000714611FC (0x000001D600000000 0x0000000000000003 0x000001D6AF6B0768 0xFFFFFFFFFFFFFFFF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x00000000714A6238 (0x000001D6AF6A8E70 0x000001D6B18A1130 0x000001D6AF6B05D8 0x0000000000000000), Tcl_ExprObj() + 0x1868 bytes(s)
0x00000000714F12F4 (0x0000000000000000 0x000001D6AF6A8E70 0x000001D6AF6B05D8 0x00000000714B460F), TclObjInterpProcCore() + 0x74 bytes(s)
0x00000000714611FC (0x000001D600000000 0x0000006000000006 0x000001D6AF6B05D8 0xFFFFFFFFFFFFFFFF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x00000000714A6238 (0x000001D6AF6A8E70 0x000001D6B299F550 0x000001D6AF6B0458 0x0000000000000000), Tcl_ExprObj() + 0x1868 bytes(s)
0x00000000714F12F4 (0x0000000000000000 0x000001D6AF6A8E70 0x000001D6AF6B0458 0x000000007152AC3C), TclObjInterpProcCore() + 0x74 bytes(s)
0x00000000714611FC (0x000001D600000000 0x000001D600000001 0x000001D6AF6B0458 0xFFFFFFFFFFFFFFFF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x00000000714A6238 (0x000001D6AF6A8E70 0x000001D6B3D611A0 0x0000000000000000 0x000001D600000000), Tcl_ExprObj() + 0x1868 bytes(s)
0x00000000714A4FEC (0x000001D6AF31CFC0 0x000001D6AF6A8E70 0x0000006004BF4D80 0x0000006004BF4D30), Tcl_ExprObj() + 0x61C bytes(s)
0x0000000071462E8D (0x000001D6B298FA50 0x0000000000000003 0x000001D6AF6A8E70 0x000001D6B39368C0), TclEvalObjEx() + 0x3AD bytes(s)
0x0000000071462ACD (0x000027A3DF7C0A94 0x000001D6AF25C3B0 0x000001D6AF31CFC0 0x00000000714F4790), Tcl_EvalObjEx() + 0x1D bytes(s)
0x00007FFB23B0075B (0x000001D6AF6A8E70 0x000001D6AF6A8E70 0x000001D6B3CBBED0 0x000000007152AC7D)
0x00000000714611FC (0x000001D600000000 0x0000000000000003 0x000001D6AF6B03E8 0xFFFFFFFFFFFFFFFF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x00000000714A6238 (0x000001D6AF6A8E70 0x000001D6B3D8F960 0x000001D6B393C3C0 0x000001D600000000), Tcl_ExprObj() + 0x1868 bytes(s)
0x00000000714A4D5D (0x000001D6B3C015F0 0x000001D6B3D7E370 0x0000000000000000 0x0000000000000001), Tcl_ExprObj() + 0x38D bytes(s)
0x000000007146D73B (0x000001D6AF6A8E70 0x0000000000000000 0x000001D6B1636080 0x000001D6AF6B0230), TclDumpMemoryInfo() + 0x5D6B bytes(s)
0x00000000714611FC (0x0000000000000000 0x000001D600000003 0x000001D6AF6B0230 0x000001D6B16363AF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x000000007146218E (0x000001D6AF6A8E70 0x000001D6B1636080 0x0000006000000003 0x000001D600000003), Tcl_EvalEx() + 0x99E bytes(s)
0x0000000071462A38 (0x000001D6AF6A8E70 0x0000000000000000 0x00007FFB1D089E1C 0x000001D6B18C1150), Tcl_Eval() + 0x38 bytes(s)
0x00007FFB1C0E2F1C (0x0000006004BF6560 0x0000000000000000 0x000001D6B18C1150 0x0000000000000000)
0x00007FFB3179A028 (0x000001D6B0FE25E0 0x0000000071510E7B 0x000001D6AF6AFD60 0x000001D6AF6A8E70), ?addOwnedOption@TclCommand@xpcl@@QEAAXPEAVOption@2@@Z() + 0x3F8 bytes(s)
0x00007FFB3179F702 (0x0000000000000000 0x000001D6AF6A8E70 0x000001D6AF6AFF60 0x000000007152AC7D), ?setResultObj@TclCommand@xpcl@@QEAAXPEAUTcl_Obj@@@Z() + 0x62 bytes(s)
0x00000000714611FC (0x0000000000000000 0x000001D600000006 0x000001D6AF6AFF58 0xFFFFFFFFFFFFFFFF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x00000000714A6238 (0x000001D6AF6A8E70 0x000001D6B3B0F140 0x000001D6AF6AFD48 0x0000000000000000), Tcl_ExprObj() + 0x1868 bytes(s)
0x00000000714F12F4 (0x0000000000000000 0x000001D6AF6A8E70 0x000001D6AF6AFD48 0x000000007152AC3C), TclObjInterpProcCore() + 0x74 bytes(s)
0x00000000714611FC (0x000001D600000000 0x000001D600000001 0x000001D6AF6AFD48 0xFFFFFFFFFFFFFFFF), Tcl_ListMathFuncs() + 0x58C bytes(s)
0x00000000714A6238 (0x000001D6AF6A8E70 0x000001D6B3C9D930 0x0000000000000000 0x000001D600000000), Tcl_ExprObj() + 0x1868 bytes(s)
0x00000000714A4FEC (0x000001D6AF31CFC0 0x000001D6AF6A8E70 0x0000006004BF7880 0x0000006004BF7830), Tcl_ExprObj() + 0x61C bytes(s)
0x0000000071462E8D (0x000001D6B298FA50 0x0000000000000003 0x000001D6AF6A8E70 0x000001D6B39368C0), TclEvalObjEx() + 0x3AD bytes(s)
[... cut ...]
0x00007FFB21F2ADE9 (0x0000000000080001 0x0000000000000000 0x0000000000000000 0x0000000000000000), ?join@HCTNativeThread@@QEAAXXZ() + 0x39 bytes(s)
0x00007FFB6F63CD70 (0x0000000000000000 0x0000000000000000 0x0000000000000000 0x0000000000000000), _o__realloc_base() + 0x60 bytes(s)
0x00007FFB728884D4 (0x0000000000000000 0x0000000000000000 0x0000000000000000 0x0000000000000000), BaseThreadInitThunk() + 0x14 bytes(s)
0x00007FFB72D3E851 (0x0000000000000000 0x0000000000000000 0x0000000000000000 0x0000000000000000), RtlUserThreadStart() + 0x21 bytes(s)
Abnormal program termination (EXCEPTION_ACCESS_VIOLATION)

bgouthamb, 2019-06-18:
Hello @Q-point,

What is your HLS tool version? As mentioned in the [ChangeLog](https://github.com/Xilinx/xfopencv#changelog), xfOpenCV 2019.1 version code cannot be synthesized with previous versions of SDx/Vivado.  2019.1 SDx tool suite is required to build/synthesize the functions in the current release.

Q-point, 2019-06-18:
Hello @bgouthamb,

I used Vivado HLS 2018.3.  You can close the issue then.

closed 2019-06-20 by bgouthamb