Xilinx/xfopencv #58: How to do pixel-wise computation?

3togo, 2019-06-20:
In previous version, the following will work but the newest version won't. 
for (int i = 0; i < HEIGHT; i++){
 for (int j = 0; j < WIDTH; j++){
      c.data[i*WIDTH+j] = a.data[i*WIDTH+j] / b.data[i*WIDTH+j];
}

bgouthamb, 2019-06-20:
@3togo 
This has been mentioned in the [ChangeLog](https://github.com/Xilinx/xfopencv#changelog) in points 10 and 11.  The new way is as follows:

`float val= a.read(i*WIDTH+j) / b.read(i*WIDTH+j);`
`c.write(i*WIDTH+j, val);`

3togo, 2019-06-20:
Can u elaborate on how to decide whether .data is a value or a structure and what  the stage of the buiild is referring to. 

Many thanks

""The datatype of xf::Mat Class pointer member 'data' , which earlier used to be always inferred as HLS arbitrary precision type, has been modified to be inferred as a Structure or a HLS arbitrary precision type based on the stage of the build.""

3togo, 2019-06-20:
The xf_axi.h seems triggering a number of errors. How to fix it? 

In file included from /opt/xfopencv/2019.1/include/common/xf_axi.h:38:0,
                 from ../../../../tb/../src/axi_sgbm_config.h:8,
                 from ../../../../tb/xf_sgbm_tb.cpp:2:
/tools/Xilinx/Vivado/2019.1/include/utils/x_hls_utils.h: In static member function ‘static ap_fixed<W, I, Q, O> xf::hls::numeric_limits<ap_fixed<W, I, Q, O> >::max()’:
/tools/Xilinx/Vivado/2019.1/include/utils/x_hls_utils.h:82:27: error: ‘numeric_limits’ is not a member of ‘hls’
             ap_int<W> m = ::hls::numeric_limits<ap_int<W> >::max();

3togo, 2019-06-21:
@bgouthamb ,

Many thanks

closed 2019-06-21 by 3togo