Xilinx/xfopencv #59: Bug on xf::resize after upgrading from 2018.3 to 2019.1

3togo, 2019-06-21:
The following function can run using xfopencv 2018.3 but not when using  2019.1. Why?
What is the meaning of "Call parameter type does not match function signature!"? 

 xf::resize <INTERPOLATION, XF_8UC1, wk_rows, wk_cols, dst_rows, dst_cols, XF_NPPC1, MAXDOWNSCALE> (wk_disp, disp);

```
Call parameter type does not match function signature!
  %wk_disp.data.V = alloca [38400 x i8], align 1
 [307200 x i8]*  call fastcc void @"xf::resize<0, 0, 160, 240, 480, 640, 1, 20>"(i32* %wk_disp.rows, i32* %wk_disp.cols, [38400 x i8]* %wk_disp.data.V, i32* %disp.rows, i32* %disp.cols, [307200 x i8]* %disp.data.V), !dbg !31033
Broken module found, compilation aborted!
Stack dump:
0.	Running pass 'Function Pass Manager' on module '/home/eli/git/bjp/sfsfsfsfsf/ips/xf_test/prj_xf_test_2019.1/solution1/.autopilot/db/a.g.1.bc'.
1.	Running pass 'Module Verifier' on function '@xf_filter_simple'
/tools/Xilinx/Vivado/2019.1/bin/rdiArgs.sh: line 267:  8517 Aborted                 (core dumped) "$RDI_PROG" "$@"
```

3togo, 2019-06-22:
The error may be because: 

A component cannot include virtual functions, function pointers, or bit fields.

3togo, 2019-06-22:
Below is the output dumps

INFO: [XFORM 203-602] Inlining function 'xf::Mat<0, 480, 640, 1>::write' into 'resizeNNBilinear<0, 160, 480, 1, 480, 640, 1, 2>' (/opt/xfopencv/2019.1/include/imgproc/xf_resize_nn_bilinear.hpp:423) automatically.
INFO: [XFORM 203-602] Inlining function 'xf::Mat<0, 480, 640, 1>::read' into 'xf::xfMat2AXIvideo<24, 0, 480, 640, 1>' (/opt/xfopencv/2019.1/include/common/xf_infra.h:70->/opt/xfopencv/2019.1/include/common/xf_infra.h:83->/opt/xfopencv/2019.1/include/common/xf_infra.h:202) automatically.
Call parameter type does not match function signature!
  %ss.data.V = alloca [76800 x i8], align 1
 [307200 x i8]*  call fastcc void @"xf::resize<1, 0, 480, 640, 160, 480, 1, 2>"(i32* %src.rows, i32* %src.cols, [307200 x i8]* %src.data.V, i32* %ss.rows, i32* %ss.cols, [76800 x i8]* %ss.data.V), !dbg !26192
Call parameter type does not match function signature!
  %ss.data.V = alloca [76800 x i8], align 1
 [307200 x i8]*  call fastcc void @"xf::resize<1, 0, 160, 480, 480, 640, 1, 2>"(i32* %ss.rows, i32* %ss.cols, [76800 x i8]* %ss.data.V, i32* %disp.rows, i32* %disp.cols, [307200 x i8]* %disp.data.V), !dbg !26193
Broken module found, compilation aborted!
Stack dump:
0.	Running pass 'Function Pass Manager' on module '/home/eli/git/bjp/sfsfsfsfsf/ips/xf_dummy/prj_xf_dummy_2019.1/solution1/.autopilot/db/a.g.1.bc'.
1.	Running pass 'Module Verifier' on function '@xf_dummy_filter'
/tools/Xilinx/Vivado/2019.1/bin/rdiArgs.sh: line 267:  7313 Aborted                 (core dumped) "$RDI_PROG" "$@"
Warning: HLS Process returned an error, skipping report opening!
Aborted!

3togo, 2019-06-27:
I have attached a testing program [https://github.com/Xilinx/xfopencv/files/3313122/prj_xf_dummy_2019.1.zip](url) above. Can anyone confirm whether the bug affect everyone or only affect me?

bgouthamb, 2019-07-01:
@3togo 
We acknowledge this issue. It is a bug with the 2019.1 Vivado HLS Synthesis flow, wherein it throws the error you  reported, when a function with _xf::Mat_ interfaces is instantiated more than once with different parameters (Basically creating different RTL for the two instances). In your case, the _xf::resize_ function is called twice with different template parameters. 

The fix has to come from Vivado HLS tool, which might take some time. However, you can consider the following workarounds:

1. Create a Class and paste the _xf:resize_ as a member function. Create new Objects of the class when multiple calls needs to be made to the function.

    example:
```
Class resizeWrapper
{
< copy the resize function definition here >

};
```

   In the accel function:

```
resizeWrapper obj1, obj2;

obj1.resize<., 1080, 1920,..> (... , ...);
obj2.resize<., 2160, 3840,..> (... , ...);

```

2. Copy the whole code of resize function into another function with a different name, like resize2. Use that function for the second call.

    example:

```
resize <., 1080, 1920,..> (... , ...);
resize2<., 2160, 3840,..> (... , ...);

```

3togo, 2019-07-02:
@bgouthamb ,

Many thanks for your reply. I did modify the code as suggested by your workaround 1. But, it makes no difference.  The errors are still there.  Below is the modified code. 

Eli

`#include "xf_dummy_accel.h"
#include "xf_config_params.h"
#include "imgproc/xf_resize.hpp"

#define MAXDOWNSCALE 2
#define INTERPOLATION   1
class JJ {
    public:
        template<int INTERPOLATION_TYPE, int TYPE, int SRC_ROWS, int SRC_COLS, int DST_ROWS, int DST_COLS, int NPC, int MAX_DOWN_SCALE>
        void resize (xf::Mat<TYPE, SRC_ROWS, SRC_COLS, NPC> & _src, xf::Mat<TYPE, DST_ROWS, DST_COLS, NPC> & _dst) {
            xf::resize<INTERPOLATION_TYPE, TYPE, SRC_ROWS, SRC_COLS, DST_ROWS, DST_COLS, NPC, MAX_DOWN_SCALE> (_src, _dst);
        }
};

void xf_dummy_filter(AXI_STREAM& src_data, AXI_STREAM& src_data_d)
{
#pragma HLS INTERFACE axis port=src_data
#pragma HLS INTERFACE axis port=src_data_d

XGRAY_SRC_IMAGE src(src_rows, src_cols);
XGRAY_SS_IMAGE ss(ss_rows, ss_cols);
XGRAY_DST_IMAGE disp(dst_rows, dst_cols);
JJ jj1, jj2;

#pragma HLS dataflow
xf::AXIvideo2xfMat(src_data, src);
//xf::resize <INTERPOLATION, XF_8UC1, src_rows, src_cols, ss_rows, ss_cols, XF_NPPC1, MAXDOWNSCALE> (src, ss);
//xf::resize <INTERPOLATION, XF_8UC1, ss_rows, ss_cols, dst_rows, dst_cols, XF_NPPC1, MAXDOWNSCALE> (ss, disp);
jj1.resize <INTERPOLATION, XF_8UC1, src_rows, src_cols, ss_rows, ss_cols, XF_NPPC1, MAXDOWNSCALE> (src, ss);
jj2.resize <INTERPOLATION, XF_8UC1, ss_rows, ss_cols, dst_rows, dst_cols, XF_NPPC1, MAXDOWNSCALE> (ss, disp);

xf::xfMat2AXIvideo(disp, src_data_d);
}
`

bgouthamb, 2019-07-02:
@3togo ,

The Class JJ has to have the whole definition of _xf::resize_ including any sub-functions it calls internally. _#include "imgproc/xf_resize.hpp"_ should be removed. 
Assuming that you use BILINEAR interpolation method, here is how the class should be defined:

```

class JJ {

    public:

	/***********************************************************************/

	template<int DEPTH, int INTERPOLATION_TYPE, int NPPC>
	void interpolatePixel(XF_CTUNAME(DEPTH,NPPC) A0, XF_CTUNAME(DEPTH,NPPC) B0, XF_CTUNAME(DEPTH,NPPC) A1, XF_CTUNAME(DEPTH,NPPC) B1, ap_ufixed<12,2> Wx, ap_ufixed<12,2> Wy, XF_CTUNAME(DEPTH,NPPC) &pixel)
	{
	#pragma HLS inline
		if(INTERPOLATION_TYPE==XF_INTERPOLATION_NN)
		{
			pixel = A0;
		}
		else
		{
			ap_ufixed<12,2> Wxy;
			ap_int<16> val0,val1,val2;
			ap_fixed<28,18> P1,P2,P3,P4;
			ap_ufixed<28,18> one_num = 1.0;

			Wxy = (Wx*Wy);    // Wx - 0.32, Wy-0.32  (Wx*Wy-0.64)  Wxy - 0.32
			val0 = (A0+B1-(B0+A1));
			val1 = (B0-A0);
			val2 = (A1-A0);

			P1 = (val0*Wxy);		// val0(16.0) * Wxy(0.32) = P1(16.32)
			P2 = (val1*Wy);		// val1(16.0) * Wy(0.32) = P2(16.32)
			P3 = (val2*Wx);		// val1(16.0) * Wx(0.32) = P3(16.32)
			P4 = (A0);					// A0(8.0) P4(8.32)

			pixel = (XF_CTUNAME(DEPTH,NPPC))((ap_fixed<32,22>)(P1  + P2 + P3 + P4));
			// to get only integer part from sum of 8.32's , right shift by 32
		}
	}
	template<int DEPTH, int INTERPOLATION_TYPE, int NPPC, int T_INDEX_INT, int NUMBEROFINPUTWORDS>
	void computeOutputPixel(XF_TNAME(DEPTH,NPPC) A0[NUMBEROFINPUTWORDS], XF_TNAME(DEPTH,NPPC) B0[NUMBEROFINPUTWORDS], ap_uint<T_INDEX_INT> initIndex, ap_uint<T_INDEX_INT> indexx[XF_NPIXPERCYCLE(NPPC)], ap_ufixed<12,2> Wx[XF_NPIXPERCYCLE(NPPC)], ap_ufixed<12,2> Wy, XF_TNAME(DEPTH,NPPC) &pixel)
	{
	#pragma HLS inline
		const int PIXELDEPTH = XF_DTPIXELDEPTH(DEPTH,NPPC);
		/*if(indexx[XF_NPIXPERCYCLE(NPPC)-1] > (initIndex+NUMBEROFINPUTWORDS*XF_NPIXPERCYCLE(NPPC)-1))
			{
				std::cout << "Insufficient number of words to resize in X" << std::endl;
				return;
			}*/
		assert((indexx[XF_NPIXPERCYCLE(NPPC)-1] < (initIndex+NUMBEROFINPUTWORDS*XF_NPIXPERCYCLE(NPPC)-1)) && "Insufficient number of words to resize in X");

		XF_PTUNAME(DEPTH) unpackX1[XF_NPIXPERCYCLE(NPPC)*NUMBEROFINPUTWORDS];
	#pragma HLS ARRAY_PARTITION variable=unpackX1 complete dim=1
		XF_PTUNAME(DEPTH) unpackX2[XF_NPIXPERCYCLE(NPPC)*NUMBEROFINPUTWORDS];
	#pragma HLS ARRAY_PARTITION variable=unpackX2 complete dim=1
		XF_PTUNAME(DEPTH) outputPixel[XF_NPIXPERCYCLE(NPPC)];
	#pragma HLS ARRAY_PARTITION variable=outputPixel complete dim=1
		for(int k=0; k<NUMBEROFINPUTWORDS; k++)
		{
	#pragma HLS UNROLL
			for(int i=0; i<XF_NPIXPERCYCLE(NPPC); i++)
			{
	#pragma HLS UNROLL
				unpackX1[k*XF_NPIXPERCYCLE(NPPC)+i] = A0[k].range((i+1)*XF_DTPIXELDEPTH(DEPTH,NPPC)*XF_CHANNELS(DEPTH,NPPC)-1,i*XF_DTPIXELDEPTH(DEPTH,NPPC)*XF_CHANNELS(DEPTH,NPPC));
				unpackX2[k*XF_NPIXPERCYCLE(NPPC)+i] = B0[k].range((i+1)*XF_DTPIXELDEPTH(DEPTH,NPPC)*XF_CHANNELS(DEPTH,NPPC)-1,i*XF_DTPIXELDEPTH(DEPTH,NPPC)*XF_CHANNELS(DEPTH,NPPC));
			}
		}
[... 360 lines cut ...]
        template<int INTERPOLATION_TYPE, int TYPE, int SRC_ROWS, int SRC_COLS, int DST_ROWS, int DST_COLS, int NPC, int MAX_DOWN_SCALE>
        void resize (xf::Mat<TYPE, SRC_ROWS, SRC_COLS, NPC> & _src, xf::Mat<TYPE, DST_ROWS, DST_COLS, NPC> & _dst) {

#pragma HLS INLINE OFF

	assert(  ((INTERPOLATION_TYPE == XF_INTERPOLATION_NN)
	        ||(INTERPOLATION_TYPE == XF_INTERPOLATION_BILINEAR)
			||(INTERPOLATION_TYPE == XF_INTERPOLATION_AREA)) && "Incorrect parameters interpolation type");
	
	if(INTERPOLATION_TYPE == XF_INTERPOLATION_AREA)
		assert( (NPC == XF_NPPC1)  && "Supported Operation Mode for Area Interpolation is XF_NPPC1. XF_NPPC2, XF_NPPC4 and XF_NPPC8 are not supported ");
	else
		assert( ((NPC == XF_NPPC8) || (NPC == XF_NPPC4) || (NPC == XF_NPPC2) || (NPC == XF_NPPC1) )  && "Supported Operation Modes XF_NPPC8, XF_NPPC4, XF_NPPC2 and XF_NPPC1");

	if(NPC == XF_NPPC2)
		assert((((_src.cols & 1) == 0) && ((_dst.cols & 1) == 0)) && "Input and ouput image widths should be multiples of 2 in NPPC2 mode");
	if(NPC == XF_NPPC4)
		assert((((_src.cols & 3) == 0) && ((_dst.cols & 3) == 0)) && "Input and ouput image widths should be multiples of 4 in NPPC4 mode");
	if(NPC == XF_NPPC8)
		assert((((_src.cols & 7) == 0) && ((_dst.cols & 7) == 0)) && "Input and ouput image widths should be multiples of 8 in NPPC8 mode");

        	resizeNNBilinear<TYPE, SRC_ROWS, SRC_COLS, NPC, DST_ROWS, DST_COLS, INTERPOLATION_TYPE, MAX_DOWN_SCALE>(_src,_dst);
        }
};
```

3togo, 2019-07-03:
@bgouthamb,
Your workaround is workable 

Many thanks 

Joe

3togo, 2019-08-30:
Simple problem happened when I call 
xf::duplicateMat and xf::equalizeHist twice in a program
using a "class" is not workable this time.

3togo, 2019-09-28:
any workaround?