Xilinx/xfopencv #61: hog

mitali0531, 2019-07-11:
having error building code in SDSoC..
shows unresolved references for 2 statements as follow.
1. cartToPolar(Dx, Dy, Mag, Angle, false);
2. img.locaterROI(wholeSize, roiofs);

p

bgouthamb, 2019-07-12:
@mitali0531 
Are you building it for reVISION platform or Base platform?
Are you building the function out of the box? If not, make sure to add the _-lopencv_core_ library in the Makefile.

mitali0531, 2019-07-14:
> @mitali0531
> Are you building it for reVISION platform or Base platform?
> Are you building the function out of the box? If not, make sure to add the _-lopencv_core_ library in the Makefile.

i am building it in base platform.
and could you be more elaborate about outside the box function thing?
thank you already for helping.

bgouthamb, 2019-07-15:
@mitali0531 
By out of the box, I mean, are you building the cloned/downloaded example as it is without modifying anything?

mitali0531, 2019-07-15:
> @mitali0531
> By out of the box, I mean, are you building the cloned/downloaded example as it is without modifying anything?

yes Sir.. i am building it as it is.

mitali0531, 2019-07-15:
Imported it directly from the xfopencv examples folder.

bgouthamb, 2019-07-16:
@mitali0531 
Can you confirm whether you have set the path to the opencv libs compiled for ARM in the Makefile, correctly?
Attaching the project here would also help us to take a closer look.

mitali0531, 2019-07-17:
how do i attach the project?
can i mail you rather than here. I need help with the solution.

bgouthamb, 2019-07-17:
@mitali0531 
Can you attach the Makefile along with the complete log file?

mitali0531, 2019-07-17:
[Makefile.docx](https://github.com/Xilinx/xfopencv/files/3402084/Makefile.docx)

here is the makefile for the code

bgouthamb, 2019-07-17:
@mitali0531 
This file seems to be corrupted. Am unable to open it.

mitali0531, 2019-07-17:
> @mitali0531
> This file seems to be corrupted. Am unable to open it.

the original file is not attaching. so i had to convert it to doc for sending.
if you could send your email id i will send the entire code to you.

bgouthamb, 2019-07-22:
@mitali0531 
I don't think sharing email-id is possible.  
You should find a way to attach here the Makefile and log file.

If you are building through GUI, make sure you have set the following libs : 

- opencv_core 

- opencv_imgcodecs 

- opencv_core 

in your project, in this section : 
_Project->right click->C/C++ build settings->SDS++ Linker ->Libraries->Libraries(-l)_

mitali0531, 2019-07-23:
that really helped

here is the make file:::

################################################################################
# Automatically-generated file. Do not edit!
################################################################################

-include ../makefile.init

RM := rm -rf _sds sd_card

# All of the sources participating in the build are defined here
-include sources.mk
-include src/subdir.mk
-include subdir.mk
-include objects.mk

ifneq ($(MAKECMDGOALS),clean)
ifneq ($(strip $(CC_DEPS)),)
-include $(CC_DEPS)
endif
ifneq ($(strip $(C_UPPER_DEPS)),)
-include $(C_UPPER_DEPS)
endif
ifneq ($(strip $(CXX_DEPS)),)
-include $(CXX_DEPS)
endif
ifneq ($(strip $(C_DEPS)),)
-include $(C_DEPS)
endif
ifneq ($(strip $(CPP_DEPS)),)
-include $(CPP_DEPS)
endif
endif

-include ../makefile.defs

# Add inputs and outputs from these tool invocations to the build variables 

# All Target
all: pre-build main-build

# Main-build Target
main-build: kuchtohho.elf

# Tool invocations
kuchtohho.elf: $(OBJS)  $(USER_OBJS)
	@echo 'Building target: $@'
	@echo 'Invoking: SDS++ Linker'
	sds++ -L"C:\Users\Meet\Desktop\opencv-3.0.0\include\opencv2" --remote_ip_cache C:/Users/Meet/workspace/ip_cache -o "kuchtohho.elf" $(OBJS) $(USER_OBJS) $(LIBS) -dmclkid 1 -mno-bitstream  -mno-boot-files  -sds-sys-config a53_linux -sds-proc a53_linux -sds-pf "zcu104" -emulation debug
	@echo 'Finished building target: $@'
	@echo ' '

# Other Targets
clean:
	-$(RM) $(CC_DEPS)$(EXECUTABLES)$(C_UPPER_DEPS)$(OBJS)$(CXX_DEPS)$(C_DEPS)$(CPP_DEPS) kuchtohho.elf
	-@echo ' '

pre-build:
	-sdsoc_make_clean Debug
	-@echo ' '

.PHONY: all clean dependents
.SECONDARY: main-build pre-build

-include ../makefile.targets

bgouthamb, 2019-07-23:
@mitali0531 
The -L path in the Makefile is pointing to the the _include_ directory of OpenCV. This is not correct. It should have the path to the _lib_ directory. 

In _Project->right click->C/C++ build settings->SDS++ Linker ->Libraries->Library Search Path(-L)_ 
add the following: _path-to-your-opencv-install-folder/lib_

Go through "Using the xfOpenCV Library on a non-reVISION Platform" section of  [UG1233](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2019_1/ug1233-xilinx-opencv-user-guide.pdf) to understand the build procedure. 

Post any further usage related issues on the [SDSoC forum](https://forums.xilinx.com/t5/SDSoC-Environment-and-reVISION/bd-p/sdsoc).

mitali0531, 2019-07-24:
Thank you so much Sir for all the help...