Xilinx/xfopencv #63: run dilation demo error on windows

Yangxiu123321, 2019-07-20:
windows 10
vivado hls 2019.1
xfopencv 2019.1
demo : xfopencv\HLS_Use_Model\Standalone_HLS_Example
simulation is ok,but synthesis is error:
ERROR: [XFORM 203-191] Unsupported pointer usage: ''  used by 'Mat<0, 1080, 1920, 1>.data.chnl.V.load'  in function  'xf::Mat<0, 1080, 1920, 1>::read40' .
ERROR: [HLS 200-70] Synthesizability check failed.
command 'ap_source' returned error code
    while executing
"source E:/sdsoc/code/vivado/Dilation/Dilation/solution1/csynth.tcl"
    invoked from within
"hls::main E:/sdsoc/code/vivado/Dilation/Dilation/solution1/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"

Yangxiu123321, 2019-07-20:
> re.cpp.tb.cpp

-std=c++14 is ok

bgouthamb, 2019-07-22:
@Yangxiu123321 

Are you following [UG1233](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2019_1/ug1233-xilinx-opencv-user-guide.pdf) doscument's chapter 4?

Make sure you have set the flag **-D__SDSVHLS__** for both the source and testbench files.

nelsoncsc, 2020-03-09:
Hi @bgouthamb, I am also facing the same issue using Vivado 2018.3. Even afterr setting the flag  -D__SDSVHLS__  the C/RTL co-simulation fails. I am using Ubuntu 18.04. However, using Vivado 2018.3 in Windows cosimulation is succesful. 
Could you please let me know any workaround for that?
Thank you

tharaka27, 2020-05-27:
I am also facing the same issue using Vivado 2018.3.  However, using Vivado 2018.3 in Windows co simulation is successful.
Could you please let me know any workaround for that?
Thank you