Xilinx/xfopencv #67: Will xfopencv support Artix-7 in near future?

3togo, 2019-08-19:
I am now planning to tailor-make a fpga board for high performance TV. I wonder whether I could use xfopencv with Atrix-7 because it is relatively cheaper than zynq series.

According to your web site,
"The xfOpenCV library is designed to work with Zynq, Zynq Ultrascale+, and Alveo FPGAs. The library has been verified on zcu102, zcu104 and U200 boards."

bgouthamb, 2019-08-21:
Hello @3togo ,

xfOpenCV is like any other HLS optimized C++ functions. It can be built for any FPGA, provided that:
 1. The FPGA has enough resources to fit the implemented function. The total resources consumed 
 would in turn affects the max. frequency of operation.
2. You got the required platform to support the build (I don't see Artix in SDx) or You take care of programming the FPGA on your own through Vivado.

You might want to take a look at Ultra96 available at comparatively lower cost.

3togo, 2019-08-25:
Many thanks

closed 2019-08-25 by 3togo