Xilinx/xfopencv #69: ip_accel_app not recognize in tb

SarahWicker, 2019-08-21:
Windows 10
Vivado 2018.3
Xfopencv 2018.3_release (tag)

I try using the hls bat and script.tcl to build the ip of dilation (HLS_Use_Model/AXI_Sample) 
i made the changes to add the xf_axi.h at the end of the xf_headers.h to correct that cvMat2AXIvideoxf<NPC1>(in_img, _src); and AXIvideo2cvMatxf<NPC1>(_dst, in_img1); weren't recognize.
Also in script.tcl i change the taget device to fit the ultra96 board, following this description: https://www.hackster.io/dhq/accelerating-image-processing-with-ultra96-caebe6

but now : ip_accel_app(_src, _dst, height, width); isn't recognize at least the error is:
```
obj/xf_dilation_tb.o: In function `main':
C:\Projekte\xfopencv\HLS_Use_Model\AXI_Sample\dilation_project\solution1\csim\build/../../../../xf_dilation_tb.cpp:78:undefined reference to ìp_accel_app(hls::stream< ap_axiu<8,1,1,1> >& ,hls::stream< ap_axiu<8,1,1,1> >& ,int ,int)'
collect2.exe: error: ld returned 1 exit status
make: *** [csim.exe] Error 1 
ERROR:[SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************

```

After looking thought the source files i saw that ip_accel_app was defined twice, inside xf_dilation_config.h :

`void ip_accel_app(hls::stream< ap_axiu<8,1,1,1> >& _src,hls::stream< ap_axiu<8,1,1,1> >& _dst,int height,int width);`

and inside xf_ip_accel_app.cpp:

```
void ip_accel_app(hls::stream< ap_axiu<8,1,1,1> >& _src,hls::stream< ap_axiu<8,1,1,1> >& _dst,int height,int width, unsigned char kernel[FILTER_SIZE*FILTER_SIZE])
{
#pragma HLS INTERFACE axis register both  port=_src
#pragma HLS INTERFACE axis register both  port=_dst

	 xf::Mat<TYPE, HEIGHT, WIDTH, NPC1> imgInput1(height,width);
	 xf::Mat<TYPE, HEIGHT, WIDTH, NPC1> imgOutput1(height,width);

#pragma HLS stream variable=imgInput1.data dim=1 depth=1
#pragma HLS stream variable=imgOutput1.data dim=1 depth=1
	#pragma HLS dataflow

	xf::AXIvideo2xfMat(_src, imgInput1);

	 dilation_accel(imgInput1,imgOutput1, kernel);

	xf::xfMat2AXIvideo(imgOutput1, _dst);

}

```
But we are not calling for this last file yet so i don't understand why it gets confused…
Doesn't anyone has an idea or already encounter this issue before?

Thank You a lot for your help :)
Sarah

bgouthamb, 2019-08-21:
Hello @SarahWicker ,

This is a reported [bug ](https://github.com/Xilinx/xfopencv/issues/27)in 2018.3 release, and was corrected through the following commit: https://github.com/Xilinx/xfopencv/commit/1cb0cf1f7ba2f4764572d1f9d26c6ff847c9f5ff#diff-42a43c4603694dd5c47cf1283374239f

You can find the corrected files here : https://github.com/Xilinx/xfopencv/tree/1cb0cf1f7ba2f4764572d1f9d26c6ff847c9f5ff/HLS_Use_Model/AXI_Sample

SarahWicker, 2019-08-22:
oh thanks i couldn't find the correct files.
Just i saw that the #include "common/xf_axi.h" was still missing in the xf_headers.h, once this corrected and the flag added to the simulation and co-simulation it worked well.
Thank you a lot for your help

closed 2019-08-22 by SarahWicker