Xilinx/xfopencv #72: ERROR: [COSIM 212-379] with boundingbox example

folixg, 2019-11-05:
I tried to get the boundingbox example working with Vivado HLS 2019.1 on Ubuntu.
C-Sim and Synthesis work fine. The Co-Simulation looks good in the beginning but the C post checking fails with a segmentation fault.

```
INFO: [Common 17-206] Exiting xsim at Tue Nov  5 22:11:24 2019...
INFO: [COSIM 212-316] Starting C post checking ...
Minimum error in intensity = 0.000000
Maximum error in intensity = 0.000000
Percentage of pixels above error threshold = 0.000000
ERROR: System recieved a signal named SIGSEGV and the program has to stop immediately!
This signal was generated when a program tries to read or write outside the memory that is allocated for it, or to write memory that can only be read.
Possible cause of this problem may be: 1) the depth setting of pointer type argument is much larger than it needed; 2)insufficient depth of array argument; 3)null pointer etc.
Current execution stopped during CodeState = ENTER_WRAPC_PC.
You can search CodeState variable name in apatb*.cpp file under ./sim/wrapc dir to locate the position.

ERROR: [COSIM 212-379] Detected segmentation violation, please check C tb.
ERROR: [COSIM 212-362] Aborting co-simulation: C TB post check failed.
ERROR: [COSIM 212-4] *** C/RTL co-simulation finished: FAIL ***
```

Please let me know if you need me to provide further information.

bgouthamb, 2019-11-08:
@folixg  Make sure the input image size matches the HEIGHT and WIDTH values defined in config_params.h

folixg, 2019-11-08:
Thank you, adapting these values fixed it.
Nevertheless, I would have expected the defined values to match with the dimensions of the provided test image. Let me know, whether I should open a PR for that.

closed 2019-11-08 by folixg