Xilinx/xfopencv #82: cvtcolor_bgr2gray sythesis issue

Q-point, 2020-03-02:
**Is there a design bug with the cvtcolor_bgr2gray function in the underlying library?
I wrapped the function in an AXI Stream and it fails. (Vivado HLS 2019.1)**

I run the simulation and it passes. I run synthesis and it fails with the error below.

This the function:

```
void ip_accel_app(hls::stream< ap_axiu<24,1,1,1> >& _src,hls::stream< ap_axiu<8,1,1,1> >& _dst,int height,int width)
{
#pragma HLS INTERFACE axis register both  port=_src
#pragma HLS INTERFACE axis register both  port=_dst
	 xf::Mat<IN_TYPE, HEIGHT, WIDTH, NPC1> imgInput1(height,width);
	 xf::Mat<OUT_TYPE, HEIGHT, WIDTH, NPC1> imgOutput1(height,width);
#pragma HLS stream variable=imgInput1.data dim=1 depth=1
#pragma HLS stream variable=imgOutput1.data dim=1 depth=1
	#pragma HLS dataflow
	xf::AXIvideo2xfMat(_src, imgInput1);
	cvtcolor_bgr2gray(imgInput1,imgOutput1);
	xf::xfMat2AXIvideo(imgOutput1, _dst);

}

```

----------------------------------------

```
NFO: [HLS 200-10] Analyzing design file 'xf_ip_accel_app.cpp' ... 
WARNING: [HLS 200-40] In file included from xf_ip_accel_app.cpp:1:
In file included from xf_ip_accel_app.cpp:1:
In file included from ./xf_cvt_config.h:6:
In file included from ../../include\common/xf_common.h:34:
../../include\common/xf_structs.h:403:33: warning: unknown attribute 'xcl_array_geometry' ignored [-Wattributes]
    DATATYPE *data __attribute((xcl_array_geometry((ROWS)*(COLS>> (xfNPixelsPerCycle<NPC>::datashift)))));
                                ^
1 warning generated.
WARNING: [HLS 200-40] In file included from xf_ip_accel_app.cpp:1:
In file included from xf_ip_accel_app.cpp:1:
In file included from ./xf_cvt_config.h:10:
../../include\imgproc/xf_cvt_color.hpp:5539:28: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                           ^~~
                           TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:5539:33: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                                ^~~
                                TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:5635:28: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                           ^~~
                           TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:5635:33: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                                ^~~
                                TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:5713:28: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                           ^~~
                           TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:5713:33: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                                ^~~
                                TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:6225:28: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                           ^~~
                           TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:6225:33: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                                ^~~
                                TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:6322:28: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                           ^~~
                           TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:6322:33: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                                ^~~
                                TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:6400:28: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                           ^~~
                           TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
../../include\imgproc/xf_cvt_color.hpp:6400:33: error: use of undeclared identifier 'iTC'; did you mean 'TC'?
_ssdm_op_SpecLoopTripCount(iTC, iTC, 0, "");
                                ^~~
                                TC
../../include\imgproc/xf_cvt_color.hpp:5496:101: note: 'TC' declared here
template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC>
                                                                                                    ^
12 errors generated.
Compilation of the preprocessed source 'xf_ip_accel_app' failed
    while executing
"source C:/HLS/xfopencv-master_2019/HLS_Use_Model/HLS_CVT_BGR_GRAY_AXIS_Example2/dilation_project/solution1/csynth.tcl
```

Tomasmed18, 2020-03-03:
Getting the exact same error right now. 
I'm using xf::rgb2gray() and xf::gray2rgb() functions.

Tomasmed18, 2020-03-03:
It seems that these functions are supported since 2019.1 xfOpenCV release.[Here's an issue asking when these type of functions would be available](https://github.com/Xilinx/xfopencv/issues/51).

I just posted a question in Xilinx's HLS forum, [here's the link](https://forums.xilinx.com/t5/High-Level-Synthesis-HLS/Synthesis-errors-with-xf-cvt-color-hpp/td-p/1080969). Let's see if we get a solution there...

Tomasmed18, 2020-03-04:
Hi @Q-point 
I ended up modifying the xf_cvt_color.hpp file, adding the template argument **iTC** in the functions where it was missing. It works now.

Q-point, 2020-03-05:
Hi @Tomasmed18 ,

So you forked the library and made underlying changes by changing TC to iTC on the template of this function? 
Is that correct?

Tomasmed18, 2020-03-05:
I kept TC and added iTC, like this:

template<int SRC_T,int DST_T,int ROWS, int COLS, int NPC, int WORDWIDTH_SRC, int WORDWIDTH_DST, int TC, **int iTC**>

Tomasmed18, 2020-03-05:
Here's the modified version, just change the file extension to .hpp

[xf_cvt_color.txt](https://github.com/Xilinx/xfopencv/files/4293319/xf_cvt_color.txt)

Q-point, 2020-03-06:
Ha, OK, now I see. Thank you. So this confirms that it's a library issue. Doing this though will overwrite the existing library file. Seems like a regression from 2018.3

nguyencanhtrung, 2021-10-04:
It's better to pull the newest Vitis Libraries to solve the problem!

lidaoyu123, 2022-03-09:
> 

thk