Xilinx/xfopencv #83: Synthesis error of meanshifttracking

Clement25, 2020-03-12:
I try to synthesize the example design "Meanshifttracking" in Vivado HLS v2019.1. The following error message occurs:
```
Unsupported pointer usage: ''  used by 'Mat<7, 1080, 1920, 1>.data.chnl.V.load'  in function  'xf::Mat<7, 1080, 1920, 1>::read69' .
```
My system is ubuntu16.04.
However, when I try to synthesize the same project in Windows system, the problem disappeeared and the whole process accomplishes sucessfully.
Could anyone shot this issue and tell me how to solve this?

Williance, 2020-03-16:
你好，在Meanshifttracking里，请问一下你进行c simulation的时候，window下Input arguments是怎么填的？

Clement25, 2020-04-28:
> 你好，在Meanshifttracking里，请问一下你进行c simulation的时候，window下Input arguments是怎么填的？

If you don't fill in the blank correctly, it will generate the standard format of input arguments.
i.e. <path-to-image> <number of objects>